ChristianTremblay/BAC0 #301: Unable to get the response to bacnet.whois()

jorgenfb, 2021-11-13:
I have a bacnet device on my network that I know responds to whois requests, I have checked with other bacnet software. However, when using BAC0s whois() function I get now answer.

This is the code I run in the repl:
```
>>> import BAC0
>>> bacnet = BAC0.lite()
2021-11-13 22:13:48,205 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2021-11-13 22:13:48,205 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2021-11-13 22:13:48,205 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-11-13 22:13:48,205 - INFO    | Starting TaskManager
2021-11-13 22:13:49,392 - INFO    | Using ip : 192.168.87.104
2021-11-13 22:13:49,395 - INFO    | Starting app...
2021-11-13 22:13:49,396 - INFO    | BAC0 started
2021-11-13 22:13:49,396 - INFO    | Registered as Simple BACnet/IP App
2021-11-13 22:13:49,400 - INFO    | Update Local COV Task started 
>>> bacnet.whois()
[]
```
I have verified using wireshark that my device actually is responding:
[image]

Using the function "whohas(object_id="analogInput:1") does work. I am also able to use read(), readMultiple, write(), writeMultiple() functions.

I appreciate any input on solving this or debug it further.

ChristianTremblay, 2021-12-11:
When defining bacnet, can you try : 

```
bacnet = BAC0.lite(ip="192.168.87.104/24") #defining subnet
```

then broadcast messages will be sent to 192.168.87.255 instead of 255.255.255.255 and I think you will have responses.

jorgenfb, 2021-12-13:
Thank you for helping me out here! 

Defining the subnet did not help with my issue. If you look at the whireshark messages my air unit actually responds after 0.2-0.3 seconds, however BAC0 does not seem to register this response. Anything I can do to debug this further?

jorgenfb, 2021-12-13:
This works. I am able to use BAC0 to talk to the device when I know its adresse. I can also use read(), readMultiple, write(), writeMultiple() functions. 

Howeverm still not able to have BAC0 detect the reply from whois :(

ChristianTremblay, 2021-12-17:
Something is weird with your network or your network interface. Can you confirm that when you specify the subnet, whois messages are sent to 192.168.87.255 ?

jorgenfb, 2021-12-28:
I ran wireshark to verify and the whois message is indeed sent to 192.168.87.255. However, the response from my device is broadcasted to 255.255.255.255. Does that mean that my device is misbehaving? Should it respond directly to the source of the whois? Or broadcast on the subnet?

The wireshark output showing the messages with their destinations:
[image]

closed 2022-03-06 by github-actions[bot]