ChristianTremblay/BAC0 #302: Using newer read_weeklySchedule & write_weeklySchedule

iamliamc, 2021-11-17:
Hey Christian, 

```
python3 -m pip freeze | grep BAC
BAC0==21.2.25
```

Working off of: 
https://bac0.readthedocs.io/en/develop/schedules.html#writing-to-the-weeklyschedule

Just attempting to read and then write back to a BACnet `schedule` `weeklySchedule` field... 

I get this error: 
`2021-11-17 17:01:42,188 - ERROR   | Error list index out of range`

On the read but it looks like the data is mostly complete? 

```
>>> bcu.bacnet_instance
Bacnet Network using ip 192.168.1.25 with device id 3056893
>>> type(bcu.bacnet_instance)
<class 'BAC0.scripts.Lite.Lite'>
>>> data = bcu.bacnet_instance.read_weeklySchedule('5:0x030000000000', 2)
2021-11-17 17:01:42,188 - ERROR   | Error list index out of range
>>> data
{'object_references': [], 'references_names': [], 'states': {0: 0, 1: 1, 2: 2, 3: 3, 4: 4, 5: 5, 6: 6, 7: 7, 8: 8, 9: 9, 10: 10, 11: 11, 12: 12, 13: 13, 14: 14, 15: 15, 16: 16, 17: 17, 18: 18, 19: 19, 20: 20, 21: 21, 22: 22, 23: 23, 24: 24, 25: 25, 26: 26, 27: 27, 28: 28, 29: 29, 30: 30, 31: 31, 32: 32, 33: 33, 34: 34, 35: 35, 36: 36, 37: 37, 38: 38, 39: 39, 40: 40, 41: 41, 42: 42, 43: 43, 44: 44, 45: 45, 46: 46, 47: 47, 48: 48, 49: 49, 50: 50, 51: 51, 52: 52, 53: 53, 54: 54, 55: 55, 56: 56, 57: 57, 58: 58, 59: 59, 60: 60, 61: 61, 62: 62, 63: 63, 64: 64, 65: 65, 66: 66, 67: 67, 68: 68, 69: 69, 70: 70, 71: 71, 72: 72, 73: 73, 74: 74, 75: 75, 76: 76, 77: 77, 78: 78, 79: 79, 80: 80, 81: 81, 82: 82, 83: 83, 84: 84, 85: 85, 86: 86, 87: 87, 88: 88, 89: 89, 90: 90, 91: 91, 92: 92, 93: 93, 94: 94, 95: 95, 96: 96, 97: 97, 98: 98, 99: 99, 100: 100, 101: 101, 102: 102, 103: 103, 104: 104, 105: 105, 106: 106, 107: 107, 108: 108, 109: 109, 110: 110, 111: 111, 112: 112, 113: 113, 114: 114, 115: 115, 116: 116, 117: 117, 118: 118, 119: 119, 120: 120, 121: 121, 122: 122, 123: 123, 124: 124, 125: 125, 126: 126, 127: 127, 128: 128, 129: 129, 130: 130, 131: 131, 132: 132, 133: 133, 134: 134, 135: 135, 136: 136, 137: 137, 138: 138, 139: 139, 140: 140, 141: 141, 142: 142, 143: 143, 144: 144, 145: 145, 146: 146, 147: 147, 148: 148, 149: 149, 150: 150, 151: 151, 152: 152, 153: 153, 154: 154, 155: 155, 156: 156, 157: 157, 158: 158, 159: 159, 160: 160, 161: 161, 162: 162, 163: 163, 164: 164, 165: 165, 166: 166, 167: 167, 168: 168, 169: 169, 170: 170, 171: 171, 172: 172, 173: 173, 174: 174, 175: 175, 176: 176, 177: 177, 178: 178, 179: 179, 180: 180, 181: 181, 182: 182, 183: 183, 184: 184, 185: 185, 186: 186, 187: 187, 188: 188, 189: 189, 190: 190, 191: 191, 192: 192, 193: 193, 194: 194, 195: 195, 196: 196, 197: 197, 198: 198, 199: 199, 200: 200, 201: 201, 202: 202, 203: 203, 204: 204, 205: 205, 206: 206, 207: 207, 208: 208, 209: 209, 210: 210, 211: 211, 212: 212, 213: 213, 214: 214, 215: 215, 216: 216, 217: 217, 218: 218, 219: 219, 220: 220, 221: 221, 222: 222, 223: 223, 224: 224, 225: 225, 226: 226, 227: 227, 228: 228, 229: 229, 230: 230, 231: 231, 232: 232, 233: 233, 234: 234, 235: 235, 236: 236, 237: 237, 238: 238, 239: 239, 240: 240, 241: 241, 242: 242, 243: 243, 244: 244, 245: 245, 246: 246, 247: 247, 248: 248, 249: 249, 250: 250, 251: 251, 252: 252, 253: 253, 254: 254}, 'reliability': 'noFaultDetected', 'priority': 8, 'presentValue': '1 (1)', 'week': {'monday': [('04:00', 1), ('21:00', 0)], 'tuesday': [('06:00', 1), ('21:00', 0)], 'wednesday': [('06:00', 1), ('21:00', 0)], 'thursday': [('06:00', 1), ('21:00', 0)], 'friday': [('06:00', 1), ('21:00', 0)], 'saturday': [], 'sunday': []}}
```
When I try to write the same data back to the device... 
```
>>> results = bcu.bacnet_instance.write_weeklySchedule('5:0x030000000000', 2, data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 170, in write_weeklySchedule
    self.send_weeklyschedule_request(request)
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 157, in send_weeklyschedule_request
    raise NoResponseFromController("APDU Abort Reason : {}".format(reason))
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : invalidDataType
```

Looking  a bit deeper around here: 
https://github.com/ChristianTremblay/BAC0/blob/d5aed211b80591a5daacffa5a645dfc51830079c/BAC0/core/functions/Schedule.py#L163

All looks reasonable...
```
>>> s = Schedule()
>>> ws = s.create_weeklySchedule(data)
>>> ws
<bacpypes.constructeddata.ArrayOf.<locals>.ArrayOf object at 0xb5e6cdf0>
>>> wpr = s.make_weeklySchedule_request('5:0x030000000000', 2, ws)
>>> wpr.dict_contents()
{'destination': '5:0x030000000000', 'apduType': 'ConfirmedRequestPDU', 'apduService': 'WritePropertyRequest', 'function': 'WritePropertyRequest', 'objectIdentifier': ('schedule', 2), 'propertyIdentifier': 'weeklySchedule', 'propertyValue': {'class': 3, 'number': 0, 'lvt': 0}, 'priority': 15}
```
Can you point me in any direction here?

ChristianTremblay, 2021-11-25:
hummm...
if I compare with the controller I have here to make tests I get: 

```
{'object_references': [('multiStateValue', 59)],
 'references_names': ['OCC-SCHEDULE'],
 'states': {'Occupied': 1, 'UnOccupied': 2, 'Standby': 3, 'Not Set': 4},
 'reliability': 'noFaultDetected',
 'priority': 15,
 'presentValue': 'Occupied (1)',
 'week': {'monday': [('00:00', 'UnOccupied'),
   ('07:00', 'Occupied'),
   ('18:00', 'UnOccupied')],
  'tuesday': [('00:00', 'UnOccupied'),
   ('07:00', 'Occupied'),
   ('18:00', 'UnOccupied')],
  'wednesday': [('00:00', 'UnOccupied'),
   ('07:00', 'Occupied'),
   ('18:00', 'UnOccupied')],
  'thursday': [('00:00', 'UnOccupied'),
   ('07:00', 'Occupied'),
   ('18:00', 'UnOccupied')],
  'friday': [('00:00', 'UnOccupied'),
   ('07:00', 'Occupied'),
   ('18:00', 'UnOccupied')],
  'saturday': [],
  'sunday': []}}
  ```

Running the same sequence I get : 
```
{'destination': '303:9',
 'apduType': 'ConfirmedRequestPDU',
 'apduService': 'WritePropertyRequest',
 'function': 'WritePropertyRequest',
 'objectIdentifier': ('schedule', 10001),
 'propertyIdentifier': 'weeklySchedule',
 'propertyValue': {'class': 3, 'number': 0, 'lvt': 0},
 'priority': 15}
```

On your side... the value of the schedule have 255 states... which is weird... It may be where BAC0 falls. Some difficulties to find the data type of those values...

Do you know how much of those states are used ?
Maybe you could try defining the states with something like
```
"states": {"Occupied": 0, "UnOccupied": 1},
```

Could it be possible that you must write to priority 8 ? (BAC0 uses @15 by default)

iamliamc, 2021-12-10:
Hey getting back to this now... 

```
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
python3 -m pip freeze | grep BAC0
BAC0==21.12.3
```

Testing again on a different device (same brand): 
```
import BAC0
from pprint import pprint

# Instantiate BAC0
b = BAC0.lite("192.168.250.25/24", ping=False, bbmdTTL=1000)

# Read get 
# 2021-12-10 20:52:12,434 - INFO    | 192.168.250.101 router to [1000]
# 2021-12-10 20:52:12,611 - ERROR   | Error list index out of range

data = b.read_weeklySchedule('1000:0xe90300000000', 2)

# Show contents
pprint(data)

data['week']['monday'] = ('07:01', 1)
# data['states'] = {0:0, 1: 1}

res = b.write_weeklySchedule('1000:0xe90300000000', 2, data)
import pdb; pdb.set_trace()
```

Output: 
```
2021-12-10 21:04:44,213 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2021-12-10 21:04:44,215 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2021-12-10 21:04:44,216 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-12-10 21:04:44,217 - INFO    | Starting TaskManager
2021-12-10 21:04:44,223 - INFO    | Using ip : 192.168.250.25
2021-12-10 21:04:44,465 - INFO    | Starting app...
2021-12-10 21:04:44,468 - INFO    | BAC0 started
2021-12-10 21:04:44,469 - INFO    | Registered as Simple BACnet/IP App
2021-12-10 21:04:44,482 - INFO    | Update Local COV Task started
2021-12-10 21:04:44,524 - INFO    | 192.168.250.101 router to [1000]
2021-12-10 21:04:44,685 - ERROR   | Error list index out of range
{'object_references': [],
 'presentValue': '0 (0)',
 'priority': 8,
 'references_names': [],
 'reliability': 'noFaultDetected',
 'states': {0: 0,
            1: 1,
            2: 2,
            3: 3,
            4: 4,
            5: 5,
            6: 6,
            7: 7,
            8: 8,
            9: 9,
            10: 10,
            11: 11,
            12: 12,
            13: 13,
            14: 14,
            ...
            ...
            250: 250,
            251: 251,
            252: 252,
            253: 253,
            254: 254},
 'week': {'friday': [('07:00', 1), ('18:00', 0)],
          'monday': [('07:00', 1), ('18:00', 0)],
          'saturday': [],
          'sunday': [],
          'thursday': [('07:00', 1), ('18:00', 0)],
          'tuesday': [('07:00', 1), ('18:00', 0)],
          'wednesday': [('07:00', 1), ('18:00', 0)]}}
File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 106, in _set_value
    if dict_schedule["states"].lower() == "analog":
AttributeError: 'dict' object has no attribute 'lower'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_bac0_schedule.py", line 20, in <module>
    res = b.write_weeklySchedule('1000:0xe90300000000', 2, data)
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 163, in write_weeklySchedule
    weeklyschedule = self.create_weeklySchedule(schedule)
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 119, in create_weeklySchedule
    for event in list_of_events
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 119, in <listcomp>
    for event in list_of_events
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 112, in _set_value
    return Integer(dict_schedule["states"][v] - 1)
KeyError: '7'
```

iamliamc, 2021-12-10:
Back on the original controller in question... 
Only two of the states are used... `{0: 0, 1: 1}` 

Same issue here: 
```
>>> results = bcu.bacnet_instance.write_weeklySchedule('5:0x030000000000', 2, data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 170, in write_weeklySchedule
    self.send_weeklyschedule_request(request)
  File "/home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py", line 157, in send_weeklyschedule_request
    raise NoResponseFromController("APDU Abort Reason : {}".format(reason))
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : invalidDataType
```
Go looking for the site-contents to play around:
```
>>> help('BAC0')
/home/debian/.local/lib/python3.7/site-packages/BAC0/__init__.py
```

`vim ~/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py`

Give myself a `import pdb; pdb.set_trace()` here: 
https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/functions/Schedule.py#L134

See what I can find about priority etc.

iamliamc, 2021-12-10:
The `2021-11-17 17:01:42,188 - ERROR   | Error list index out of range`

I get is because the `object_references` in my controller is an empty list doesn't seem so important...
```
IndexError: list index out of range
> /home/debian/.local/lib/python3.7/site-packages/BAC0/core/functions/Schedule.py(210)read_weeklySchedule()
-> first_obj_id = object_references[0]
(Pdb) object_references
[]

```

iamliamc, 2021-12-10:
Looks a bit like that's what's loading up the states `range(255)` stuff... 

https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/functions/Schedule.py#L235
```
        except Exception as error:
            self._log.error("Error {}".format(error))
            # Not used in device, not linked...
            _state_text = range(255)
            schedule["object_references"] = []
            schedule["references_names"] = []
```
https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/functions/Schedule.py#L250

```        
        elif _state_text != "analog":
            for i, each in enumerate(_state_text):
                schedule["states"][each] = i + offset_MV
````

Sooo maybe if I don't try to write back a bunch of states that didn't exist in the system I won't get the permission denied lets see...

iamliamc, 2021-12-10:
So right here in `_set_value`
https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/functions/Schedule.py#L104

With the fail over behavior from before it looks like we get `-1` values which I guess shouldn't be allowed...

```
(Pdb) Integer(dict_schedule["states"][v] - 1)
<bacpypes.primitivedata.Integer object at 0xb4bee050>
(Pdb) Integer(dict_schedule["states"][v] - 1).value
-1
```

iamliamc, 2021-12-10:
Looking right here: https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/functions/Schedule.py#L121

```
check = [_daily_schedule[0].dict_contents(), _daily_schedule[1].dict_contents()]
{'time': '04:00', 'value': <bacpypes.primitivedata.Integer object at 0xb6132130>} {'time': '21:00', 'value': <bacpypes.primitivedata.Integer object at 0xb4c55fb0>}
(Pdb) check[1]['value']= check[1].get('value').value
(Pdb) check[0]['value']= check[0].get('value').value
(Pdb) check
[{'time': '04:00', 'value': 0}, {'time': '21:00', 'value': -1}]
```

So it tried to write a schedule that starts with state `0 (1 - 1)` and ends with -1 `(0 - 1)`

iamliamc, 2021-12-10:
Ok so I got it to work I added 1 minute to the monday schedule.... 
```
 'week': {'friday': [('06:00', 1), ('21:00', 0)],
          'monday': [('04:01', 1), ('21:00', 0)],
          'saturday': [],
          'sunday': [],
          'thursday': [('06:00', 1), ('21:00', 0)],
          'tuesday': [('06:00', 1), ('21:00', 0)],
          'wednesday': [('06:00', 1), ('21:00', 0)]}}
```

I was continuing to get:
```
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : invalidDataType
```

Until I wrapped the value in:
```
from bacpypes.basetypes import (
    Enumerated,
)
```

iamliamc, 2021-12-10:
This is what fixed it for me -- what was the `-1` there to do...?
```
        def _set_value(v):
            try:
                if dict_schedule["states"].lower() == "analog":
                    return Real(v)
            except AttributeError:
                if dict_schedule["states"] == ["inactive", "active"]:
                    return Integer(dict_schedule["states"][v])
                else:
                    return Enumerated(dict_schedule["states"][v])
```

iamliamc, 2021-12-11:
I'll try and put a PR together at some point... just recollecting this issue started somewhere: 
https://github.com/ChristianTremblay/BAC0/issues/225

At that point I had written my own wrapper around BAC0 to deal with schedules directly through the basic `.read` and `.write` methods. 

Am I right in understanding all this code is about supporting writing `multistateValue`, `analog` and `binary` valued schedule types...?

ChristianTremblay, 2021-12-11:
Looks like a good catch.... the controller I have (Johnson Controls) accepts `Integer` when dealing with MV, but yours seems to require Enumerated.

That said, looks like a better approach.. I'll try on my side with Enumerated.

iamliamc, 2022-01-04:
It's quite strange this issue seems to be fixed: 
[#225](https://github.com/ChristianTremblay/BAC0/issues/225)

So while working on a PR I discovered that my former approach works again:

Passing a data object like this:
```
data = [
    {'daySchedule': [{'time': '00:00:00', 'value': 1}, {'time': '18:00:00', 'value': 0}]}, 
    {'daySchedule': [{'time': '01:01:00', 'value': 1}, {'time': '18:00:00', 'value': 0}]}, 
    {'daySchedule': [{'time': '02:00:00', 'value': 1}, {'time': '18:00:00', 'value': 0}]}, 
    {'daySchedule': [{'time': '03:00:00', 'value': 1}, {'time': '18:00:00', 'value': 0}]}, 
    {'daySchedule': [{'time': '03:00:00', 'value': 1}, {'time': '18:00:00', 'value': 0}]}, 
    {'daySchedule': []}, 
    {'daySchedule': []}
]
```

To the function `dict_to_weeklyschedule` wraps all that data up as `bacpypes` objects... 
```
    def dict_to_weeklyschedule(self, val):
        return ArrayOf(DailySchedule)([self.dict_to_dailyschedule(x) for x in val])

    def dict_to_dailyschedule(self, val):
        return DailySchedule(
            daySchedule=[self.dict_to_timevalue(x) for x in val["daySchedule"]]
        )

    def dict_to_timevalue(self, val):
        return TimeValue(
            time=self.to_time_tuple(val["time"]), value=self.to_enumerated(val["value"])
        )
        
        ```python
    def to_enumerated(self, val):
        return Enumerated(val)
```

```
bac_data = self.dict_to_weeklyschedule(data)
bac0_instance.write(
    Bac0Wrapper(
        network,
        object_type,
        object_id,
        "weeklySchedule",
        bac_data,
    )
)
```

All Bac0Wrapper does is make an argument for the `.write` function (in the place of the standard `str` `f"{network} {object_type} {object_id} {field} {new_value}"`) that can contain complex data types (i.e. `bacpypes` objects)... it gets passed in instead of the `str` which eventually has `.split` called on it and provides the `bacpypes` data to your interface. 
```
class Bac0Wrapper:
    def __init__(self, *arguments):
        self.args = list(arguments)

    def split(self):
        return self.args
```

```

ChristianTremblay, 2022-01-07:
The write function is mostly based on the example from bacpypes repo. When I started to develop BAC0 I worked a lot with that, making string that would be understood by bacpypes but for complex cases, creating a string is ... hard.
This is why I prefer, when possible, to create a request myself. This is what function are doing. 

But if your string is created correctly, it will work. Because bacpypes will take that string, and make a request with it.

You may fall into some edge cases though. If the conversion goes bad.

HEre you can see how the request is built : 
```
    def make_weeklySchedule_request(self, destination, object_instance, weeklySchedule):
        request = WritePropertyRequest(
            objectIdentifier=("schedule", object_instance),
            propertyIdentifier="weeklySchedule",
        )

        address = Address(destination)
        request.pduDestination = address
        request.propertyValue = Any()
        request.propertyValue.cast_in(weeklySchedule)
        request.priority = 15
        return request
```

I then take that I shoot that to the IOCB

ChristianTremblay, 2022-01-07:
If it's possible for you, I replaced Integer by Enumerated in the last develop iteration. fbbc0d9

closed 2022-03-16 by github-actions[bot]