ChristianTremblay/BAC0 #305: Points.py: self.value can be a dict

carsten-AEI, 2021-12-17:
I am just learning how to use BAC0 but found that for at least one of our "industrial" BACnet devices it sometimes errors out when `self.value` in `core/devices/Points.py` happens to be a `dict`:

```
Traceback (most recent call last):
  File "./bacnet_discover.py", line 47, in <module>
    print(p)
  File "/usr/local/lib/python3.7/dist-packages/BAC0/core/devices/Points.py", line 729, in __repr__
    val = float(self.value)
TypeError: float() argument must be a string or a number, not 'dict'
```

inserting a print block almost similar to the return statement for this method just before this line
```
print("{}/{} : {} {}".format(
    self.properties.device.properties.name,
    self.properties.name,
    self.value,
    self.properties.units_state,
))
```
returns `opk_KM1/Bac_M_Ampere_1_KM1 : {'file_presentValue': 46.0} amperes` or `opk_KM1/BAC.V1.MOD.RegDisT : {'file_presentValue': 40.099998474121094} degreesCelsius`, ...

What I wonder is why this seems to be quite random, on another run, it terminates at a different data point, e.g. `opk_KM1/Bac_Aqua_Leistung_KM1 : {'file_presentValue': 2556.0} kilowatts`

BAC0 version is the latest released one, i.e. `2021-12-17 09:15:08,864 - INFO    | Starting BAC0 version 21.12.03 (Lite)`

Obviously, I could code around this and just use the value of the dict (maybe even taking only the "first" key/value pair if there were many), but so far, I do not know why a dict could end up there. Do you?

Anything I should provide you with to help?

Test program I currently use
```python
import BAC0
BAC0.log_level('info')
bacnet = BAC0.connect(ip='192.168.1.1/24')
bacnet.whois('*:*')
devices = []
for dev in bacnet.devices:
    devices.append(BAC0.device(dev[2], dev[3], bacnet, poll=600))
for dev in bacnet.registered_devices:
    for p in dev.points:
        print(p)
```

ChristianTremblay, 2021-12-17:
a dict ? wow... pretty sure it's not conforming to standard. Maybe @JoelBender could confirm...

carsten-AEI, 2021-12-17:
yeah, I assume something is off here. Formerly, we sometimes used `bacnet-stack` (v0.8.3) and for some of these data points above the returned values look ok to me:
```
BACNET_IFACE=enp3s0 bacnet-stack-0.8.3/bin/bacrpm 2024 2 10 85 2 29 85 2 53 85 2 16 85 2 17 85 2 35 85 2 36 85
analog-value #10
{
    present-value: 61.000000
}
analog-value #29
{
    present-value: 61.000000
}
analog-value #53
{
    present-value: 39.900002
}
analog-value #16
{
    present-value: 464.000000
}
analog-value #17
{
    present-value: 2481.000000
}
analog-value #35
{
    present-value: 454.000000
}
analog-value #36
{
    present-value: 2430.000000
}
```
(being a Python n00b does not really help debugging this, sorry :( )

carsten-AEI, 2021-12-17:
OK, a bit more progress from my side (if I produce too much text, please let me know ;-))

The problem seems to be coming from casting within `def read` in `core/io/Read.py`, e.g.

```
DEBUG: Retrieve value
DEBUG: first if, returning cached value 16.299999237060547
opk_KM1/BAC.V1.MOD.Power_kW : 16.30 kilowatts
DEBUG: Retrieve value
DEBUG: network read for 192.168.2.9 analogValue 56
2021-12-17 14:33:53,008 - INFO    | iocb                 <bacpypes.iocb.IOCB(58) instance at 0x7f3b7f565ef0>
2021-12-17 14:33:53,029 - INFO    | <bacpypes.constructeddata.Any object at 0x7f3b7f62a978>
2021-12-17 14:33:53,029 - INFO    | file
2021-12-17 14:33:53,030 - INFO    | presentValue
2021-12-17 14:33:53,030 - INFO    | {'file_presentValue': 23516.0}
DEBUG: result {'file_presentValue': 23516.0}
opk_KM1/BAC.V1.MOD.Act_RPM : {'file_presentValue': 23516.0} cyclesPerMinute
DEBUG: Retrieve value
DEBUG: first if, returning cached value {'file_presentValue': 23516.0}
Traceback (most recent call last):
  File "./test.py", line 10, in <module>
    print(p)
  File "/usr/local/lib/python3.7/dist-packages/BAC0/core/devices/Points.py", line 739, in __repr__
    val = float(self.value)
TypeError: float() argument must be a string or a number, not 'dict'
```

where I sprinkled a couple of `log.info` messages into the mix (`BAC0.log_level('debug')` did not create more output for me):
```
            if not datatype:
                # raise TypeError("unknown datatype")                                                                                       
                value = cast_datatype_from_tag(
                    apdu.propertyValue,
                    apdu.objectIdentifier[0],
                    apdu.propertyIdentifier,
                )
                self._log.info(apdu.propertyValue)
                self._log.info(apdu.objectIdentifier[0])
                self._log.info(apdu.propertyIdentifier)
                self._log.info(value)
```

thus `cast_datatype_from_tag` seems to create the dict from `apdu.objectIdentifier[0])` and `apdu.propertyIdentifier`.

But why, that's something I don't get just yet.

carsten-AEI, 2021-12-17:
When this error occurs, so far it always enters the `true` branch of https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/io/Read.py#L816 e.g. with `tag` being integer `4` and `datatype` being `<class 'bacpypes.primitivedata.Real'>` then line 825 will create the `dict` out of it.

However, I'm at a loss why and why no-one else noted this before me (or why https://github.com/ChristianTremblay/BAC0/blob/11a4cb44f287974117472b1514f339cf526f4d96/BAC0/core/io/Read.py#L147 returns no datatype).

ChristianTremblay, 2021-12-17:
I just don't see why it gives a dict...

I mean... when doing your read request... you're telling the controller : 

`give me the present value of analogValue 56`

No reason for it to return `{'file_presentValue': 23516.0}`

In the Standard...

> 12.4.4 - Present value
> This property, of type REAL, indicates the current value, in engineering units, of the analog value.

What is that device ?
BTL certified ?

carsten-AEI, 2021-12-19:
I ran wireshark on the connection yesterday and I do see the wrong `file` reply there as well - thus sorry for thinking the problem may be in the "casting" method.

I will contact the cooling device's company on Monday about this -  based on the MAC address the device (modbus to bacnet gateway?) seems to be from Saia PCD but I do not know if the programming/details are made by this company or the company of our cooling device.

I can keep you posted if you want to or we could close this issue as the problem looks to be external to BAC0.

JoelBender, 2021-12-19:
Please keep us posted here.  For more information to give to the vendor, read the **Vendor_Name** (a.k.a. vendorName), **Vendor_Identifier**, **Model_Name**, **Firmware_Revision**, and **Application_Software_Version** from the device object and that will help the vendor resolve the problem, perhaps it's a simple as installing new firmware.  Some devices have components from multiple vendors, so the OEM content might be different than face-plate descriptions, maybe the list of [BACnet Vendor Identifiers](http://www.bacnet.org/VendorID/BACnet%20Vendor%20IDs.htm) can help.

carsten-AEI, 2022-09-29:
Sorry for this very late update - it took until August to sort this eventually out with the vendor, I some relevant pieces of hardware had quite the lead time. After the vendor updated the hardware and installed a newer firmware, BAC0 does not show any errors anymore when scanning this device.

Please close this issue again as this was a problem outside of this project :grinning: and thanks a bunch for BAC0 in the first place!