ChristianTremblay/BAC0 #306: how to read MSTP device when the bacnet mstp router UDP port is 47811

JerryXinhui, 2021-12-29:
hello,  the BAC0 use 47808 as default UDP port. I know for IP device, if the UDP port is 47811, I can write as following to read device object. 

```
localIP = "192.168.251.10/24:47811"
bacnet = BAC0.connect(localIP)
value = bacnet.read("192.168.251.103:47811 analogInput 12 presentValue")
print(value)
```

But for the MSTP device, if the bacnet mstp router UDP port is 47811, how to write my code? Thanks!  
 I tried as followings , not work well.  
```
localIP = "192.168.251.10/24:47811"
bacnet = BAC0.connect(localIP)
value = bacnet.read("1:5 analogInput 12 presentValue")
print(value)
```

ChristianTremblay, 2022-01-03:
Good question. I'll have to test

ChristianTremblay, 2022-01-07:
```
import BAC0
bacnet = BAC0.lite(ip='192.168.211.54/24:47811')
bacnet.discover()

```
From there, the port is known by bacpypes and eveything will use this port...

```
In [5]: bacnet = BAC0.lite(ip='192.168.211.54/24:47811')
2022-01-06 21:32:11,120 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-01-06 21:32:11,121 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-01-06 21:32:11,121 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-01-06 21:32:11,122 - INFO    | Using ip : 192.168.211.54:47811
2022-01-06 21:32:11,241 - INFO    | Starting app...
2022-01-06 21:32:11,242 - INFO    | BAC0 started
2022-01-06 21:32:11,242 - INFO    | Registered as Simple BACnet/IP App
2022-01-06 21:32:11,267 - INFO    | Update Local COV Task started

In [6]: bacnet.discover()
2022-01-06 21:32:15,437 - WARNING | 192.168.211.3:47811 Rejected message to network (reason : It is an unknown network layer message)
2022-01-06 21:32:17,231 - INFO    | 192.168.211.3:47811 router to [303]
2022-01-06 21:32:19,174 - INFO    | Addr : 192.168.211.3:47811
2022-01-06 21:32:19,237 - INFO    | 192.168.211.3:47811 routing table
2022-01-06 21:32:19,237 - INFO    |     303 1 bytearray(b'')
2022-01-06 21:32:19,237 - INFO    |     211 2 bytearray(b'')
2022-01-06 21:32:21,191 - INFO    | Found those networks : {303}
2022-01-06 21:32:21,191 - INFO    | Discovering network 303
Out[6]: 
[('303:7', 5007),
 ('303:8', 72008),
 ('303:9', 5209),
 ('303:10', 60010),
 ('303:12', 5012),
 ('303:4', 5204)]

In [17]: bacnet.read('303:4 analogInput 1019 presentValue')
Out[17]: 150534.484375
```

JerryXinhui, 2022-01-10:
yes, I tried this and works well, thanks!

closed 2022-01-10 by JerryXinhui