ChristianTremblay/BAC0 #310: Handling disconnected devices

ChristianTremblay, 2022-01-13:
This needs to be improved. @kushan-26 reported multiple pings flooding the network for no good reasons... 
The idea was to ping the device every 5 minutes to see if it's back online.

ChristianTremblay, 2022-01-13:
We could use this : https://github.com/ChristianTremblay/BAC0/discussions/309 to discuss ideas

kushan-26, 2022-01-14:
I'd like to clear out one thing first, I see all the updates are done to the Lite.py script and not much changed in the Complete.py script. Are you trying to move away from having two connectivity options based on installed packages?

ChristianTremblay, 2022-01-14:
Connect is a subclass of lite with a web server. 

So technically everything I put in lite will work in "complete" (which would be a better name than connect... if I could come back in time....)

And often, connect reverts to lite when you don't have Bokeh or flask installed. 

But I admit, I don't use the web version a lot... I'm not good enough to make things look good. Also, Bokeh server to feed real time value as trend was good... but now that I can link to influxDb and use the power of other tools... like grafana... feels like I would work a lot to make something that is way less powerful. 

What do you think ?

kushan-26, 2022-01-15:
The web interface is probably the feature I use the least too. But it's a good feature to have when testing standalone networks. So, I think what's available right now with the webserver should be sufficient to visualize data in case there's no internet connection. 

Is there a difference between a device being 'offline' or 'disconnected' in BACnet?

ChristianTremblay, 2022-02-02:
Good question.... I may have used both terms ... I don't know if the standard is strict about the use of one term over the other...

kushan-26, 2022-04-04:
Commenting to keep the issue alive. No progress was made on handling the issue. Further testing is needed.

closed 2022-06-11 by github-actions[bot]