ChristianTremblay/BAC0 #311: create_proprietary_object does not support other bacpypes data types

rhinomikey2896, 2022-01-13:
When trying to create the separate classes for proprietary objects the create_proprietary_object command does not allow for any other data type other than "primitive". The error that I receive is:

props = [
    Property(v["obj_id"], v["primitive"], mutable=v["mutable"])
KeyError: 'primitive'

When I looked at the package I found that "props" is has "primitive" as the only option, however, the Property class of bacpypes does allow for other data types specifically constructed data. To confirm that it was the different data type I commented out the line with the different data type and tried again and there was no issues with any of the other properties in the dictionary.

Is this a limitation of the create_proprietary_object method or am I missing something with the command?

Example Code:
from BAC0.core.proprietary_objects.object import create_proprietary_object
from MyPackage.MyModule import MyDeviceProprietaryProperties

create_proprietary_object(MyDeviceProprietaryProperties)

# In MyDeviceProprietaryProperties class
from bacpypes.primitivedata import (Real, Boolean, CharacterString, Enumerated, Unsigned, Atomic, Date, Time, OctetString,)
from bacpypes.object import (Object, DeviceObject, AnalogValueObject, AnalogInputObject, AnalogOutputObject, BinaryValueObject, 
        BinaryInputObject, BinaryOutputObject, Property, register_object_type,)
from bacpypes.constructeddata import (Any, Array, ArrayOf, List, ListOf, Choice, Element, Sequence)

MyDeviceProprietaryProperties= {
    "name": "MyDeviceProprietaryProperties",
    "vendor_id": 0,
    "objectType": "binaryValue",      # Yes the object type in question is of type binaryValue
    "bacpypes_type": BinaryValueObject,
    "properties": {
        "My_Constructed_Data": {"obj_id": 1234, "constructeddata": Array, "mutable": False},
    },
}

Thanks in advance!

ChristianTremblay, 2022-01-14:
"primitive" could simply be renamed "datatype".... it's only an index name in the dictionary used to create the property.

Please have a look to the develop branch and see if it works for you

rhinomikey2896, 2022-02-01:
I tried the develop branch using the "datatype" string instead of "primitive" and I am getting the same key error for "datatype" and then again for "primitive" like before.

rhinomikey2896, 2022-02-02:
So after stepping through the code a bit more I noticed that if you have any "primitive" types listed before the "datatype" line in the dictionary that the try/except statement that was added is what causes the error because the for loop does not recheck if the value is "primitive" or "datatype" and it just assumes that it is "primitive".

ChristianTremblay, 2022-02-02:
Can you be more specific ?

Are you talking about the value itself or the dict key ? 
The tests here seemed to cover your case : https://github.com/ChristianTremblay/BAC0/commit/2733641d19517b9e91de1433f3220a4136e381e0

Thanks

rhinomikey2896, 2022-02-04:
In this section of code from the create proprietary objects file: 

    try:
        props = [
            Property(v["obj_id"], v["datatype"], mutable=v["mutable"])
            for k, v in params["properties"].items()
        ]
    except KeyError:
        props = [
            Property(v["obj_id"], v["primitive"], mutable=v["mutable"])
            for k, v in params["properties"].items()
        ]

This try except only works if, when creating the class for a proprietary object, the properties key in the dictionary does not have any values what are of the primitive type. I'll include an example below: 

MyDeviceBinaryValueObject = {
    "name": "MyDeviceBinaryValueObject ",
    "vendor_id": 0,
    "objectType": "binaryValue",
    "bacpypes_type": BinaryValueObject,
    "properties": {
        "Device_Point_Prop": {"obj_id": 1234, "primitive": CharacterString, "mutable": True},
        "Device_Time_Stamps": {"obj_id": 5678, "datatype": ArrayOf(Real), "mutable": True},
    },
}

The example above should fail as the try except will go into the except clause and stay there until the for loop is complete and it will fail as the next property is a datatype instead of a primitive. I have tested it be removing the primitive and the develop branch does work but with both primitive and datatype in the order shown above the code fails and returns a key error for both datatype as well as primitive.

closed 2022-06-13 by github-actions[bot]