ChristianTremblay/BAC0 #312: Sending Binary and Multistate "value" as an integer to influxdb

kushan-26, 2022-01-17:
Hello Christian,

I noticed that the binary and multistate values including the string_value are being sent as a string to influxdb. These point types cannot be plotted for a specific time interval to see a trend or do any aggregation other than "Last" due to this reason. I have twiddle with the code to see if I can get the "value" part as an integer but failed to do so.

Futhermore, I noticed that analogue values show two tables for "value and "string_value" in influxdb raw data. Binary and multistate shows up in a single table. Think the issue lies within how these tables are created or queried in influxdb. I guess the question is how can we create two tables for binary and multistate. 
[image]

Can you think of any solution for this?

ChristianTremblay, 2022-01-18:
That is not the intent. If you look here : 
```
def clean_value(self, object_type, val, units_state):
        if "analog" in object_type:
            _string_value = "{:.3f} {}".format(val, units_state)
            _value = val
        elif "multi" in object_type:
            _string_value = "{}".format(val.split(":")[1])
            _value = "{}".format(val.split(":")[0])
        elif "binary" in object_type:
            try:
                _string_value = "{}".format(units_state[int(val.split(":"[0]))])
            except:
                _string_value = "{}".format(val.split(":")[1])
            _value = "{}".format(val.split(":")[0])
        else:
            _string_value = "{}".format(val)
            _value = val
        return (_value, _string_value)
```
I create 2 distinct fields, a string_value that is meant to be used by tools that can work with strings... (last value presenting the actual state for example)... and value, which hold the numeric part of the information.

[image]

If filtering by _field : value
[image]

Here you can see examples where I use the string value
[image]

and here
[image]

kushan-26, 2022-01-18:
I do understand the string value and its usage. I'm actually talking about the numeric part of the binary and multistate points. The numeric part is identified as a string in influxdb. If you select a binary point -> value field and try to use the 'mean' aggregate function, you'd see the following error: 'unsupported input type for mean aggregate: string'. Shouldn't that be an integer or double type value?
[image]

Also, if you do the same filter (binary point -> value) with the 'last' aggregate function and plot a graph, output the trend is sort of broken because of string numeric values.
[image]

I tried changing the following lines in bold to convert the string to integer but that broke the _value and only received the '_string_value' part to the database.
        elif "multi" in object_type:
            _string_value = "{}".format(val.split(":")[1])
            _value = **"{}".format(val.split(":")[0])** -> int(val.split(":")[0])
        elif "binary" in object_type:
            try:
                _string_value = "{}".format(units_state[int(val.split(":"[0]))])
            except:
                _string_value = "{}".format(val.split(":")[1])
            _value = **"{}".format(val.split(":")[0])** -> int(val.split(":")[0])

Hope this explains the issue I'm trying to point out.

kushan-26, 2022-01-18:
@Christian - Actually what I changed in the script did work. It was just an issue with influxdb not being able to handle two data types under the same record in the same bucket. I created a new Bucket and was able to receive both string_value and numeric value as a float type. I've created a pull request with changes to the script.

Just needed a fresh day to think different...lol

Thanks for your help!

ChristianTremblay, 2022-01-20:
This bug did affect my own DB here and I also have string values instead of `integer`... but there is always a solution !

```
from(bucket: "BAC0")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "Device_5/binaryInput:7")
  |> filter(fn: (r) => r["_field"] == "value")
  |> toInt()
  |> aggregateWindow(every: v.windowPeriod, fn: mean, createEmpty: false)
  |> yield(name: "mean")
```

`toInt()` convert all `_value` column to int

flux is cool

kushan-26, 2022-01-20:
That is pretty cool. I did not think about it that way.

closed 2022-01-20 by kushan-26