ChristianTremblay/BAC0 #315: problems with trend and BACpypes openweatherserver sample (maybe a missing service?)

epaulson, 2022-01-28:
I've been experimenting a bit with BAC0, using the BACpypes samples/OpenWeatherServer.py as a sample device, on a 2-node private VM network. 

It didn't work out of the box at all - BAC0 seems to require that device support the ReadPropertyMultiple service, which the BACpypes BIPSimpleApplication doesn't include, but that was an easy fix. 

However, I'm not sure why I can't BAC0 trends to work. I think I'm doing it right according to the docs, but if I add a trend, a few seconds later BAC0 removes the trend, and I never see anything in the Bokeh web interface. Here's a transcript:
```
>>> import BAC0
>>> bacnet = BAC0.connect('10.0.2.8/24')
2022-01-28 14:03:41,849 - INFO    | Starting BAC0 version 21.12.03 (Complete)
2022-01-28 14:03:41,849 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-01-28 14:03:41,849 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-01-28 14:03:41,849 - INFO    | Starting TaskManager
2022-01-28 14:03:41,850 - INFO    | Using ip : 10.0.2.8
2022-01-28 14:03:41,868 - INFO    | Starting app...
2022-01-28 14:03:41,870 - INFO    | BAC0 started
2022-01-28 14:03:41,871 - INFO    | Registered as Simple BACnet/IP App
2022-01-28 14:03:41,874 - INFO    | Update Local COV Task started
2022-01-28 14:03:43,037 - INFO    | Server started : http://10.0.2.8:8111
 * Serving Flask app 'BAC0.web.FlaskServer' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses.
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://10.0.2.8:8111/ (Press CTRL+C to quit)
2022-01-28 14:03:47,041 - INFO    | Found those networks : set()
2022-01-28 14:03:47,041 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
>>> openweather = BAC0.device('10.0.2.7/24', 599, bacnet)
2022-01-28 14:04:26,857 - INFO    | Changing device state to DeviceDisconnected'>
2022-01-28 14:04:26,866 - INFO    | Changing device state to RPMDeviceConnected'>
2022-01-28 14:04:26,877 - INFO    | Device 599:[WeatherServer] found... building points list
2022-01-28 14:04:26,910 - INFO    | Ready!
2022-01-28 14:04:26,910 - INFO    | Device defined for normal polling with a delay of 10sec
2022-01-28 14:04:26,910 - INFO    | Polling started, values read every 10 seconds
>>> openweather.points
[WeatherServer/$.clouds.all : 0.00 percent, WeatherServer/$.main.humidity : 58.00 percentRelativeHumidity, WeatherServer/$.main.pressure : 1026.00 hectopascals, WeatherServer/$.main.temp : -8.91 degreesCelsius, WeatherServer/$.main.temp_max : -7.66 degreesCelsius, WeatherServer/$.main.temp_min : -10.88 degreesCelsius, WeatherServer/$.visibility : 10000.00 meters, WeatherServer/$.wind.deg : 297.00 degreesAngular, WeatherServer/$.wind.speed : 1.34 metersPerSecond, WeatherServer/$.sys.sunrise : 2022-01-28 07:16:49, WeatherServer/$.sys.sunset : 2022-01-28 17:04:07]
>>> openweather['$.main.temp']
WeatherServer/$.main.temp : -8.91 degreesCelsius
>>> openweather['$.main.temp'].history
2022-01-28 14:04:26.903443-06:00   -8.91
2022-01-28 14:04:42.515887-06:00   -8.91
2022-01-28 14:04:52.530377-06:00   -8.91
2022-01-28 14:05:02.539441-06:00   -8.91
2022-01-28 14:05:12.551985-06:00   -8.91
Name: WeatherServer/$.main.temp, dtype: float64
>>> openweather['$.main.temp'].chart()
>>> 10.0.2.8 - - [28/Jan/2022 14:05:43] "GET / HTTP/1.1" 200 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/css/bootstrap.min.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/css/animate.min.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/css/paper-dashboard.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/css/demo.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/css/themify-icons.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:43] "GET /static/assets/js/jquery-1.10.2.js HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /trends HTTP/1.1" 200 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/css/bootstrap.min.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/css/animate.min.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/css/paper-dashboard.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/css/demo.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/css/themify-icons.css HTTP/1.1" 304 -
10.0.2.8 - - [28/Jan/2022 14:05:48] "GET /static/assets/js/jquery-1.10.2.js HTTP/1.1" 304 -
2022-01-28 14:05:52,328 - INFO    | Removing WeatherServer/$.main.temp from trends
>>> 
```

Is there another BACnet service I need to add to the OpenWeatherServer app, or is there something else I'm doing wrong?

Thanks!

ChristianTremblay, 2022-02-02:
humm.... it's been a long time I checked up the Bokeh side of BAC0... 
Since I configured the influxDB option, the short-term feature of Bokeh lost a lot of interest.

That said, I would be curious to see the code you use... maybe I can see something.

epaulson, 2022-02-02:
My server is almost stock BACpypes OpenWeatherServer, with just a few changes so BAC0 will talk to it. (You can ignore the change for the city, I just wanted to see my hometown):

```
diff --git a/samples/OpenWeatherServer.py b/samples/OpenWeatherServer.py
index 6183c0d..2bd3d2b 100644
--- a/samples/OpenWeatherServer.py
+++ b/samples/OpenWeatherServer.py
@@ -23,6 +23,8 @@ from bacpypes.basetypes import DateTime
 from bacpypes.object import AnalogValueObject, DateTimeValueObject
 
 from bacpypes.app import BIPSimpleApplication
+from bacpypes.service.device import DeviceCommunicationControlServices
+from bacpypes.service.object import ReadWritePropertyMultipleServices
 from bacpypes.local.device import LocalDeviceObject
 
 # some debugging
@@ -39,7 +41,7 @@ APPUNITS = os.getenv("APPUNITS", "metric")
 APPINTERVAL = int(os.getenv("APPINTERVAL", 5)) * 60 * 1000
 
 # request parameters are passed to the API, id is the city identifier
-request_parameters = {"id": 7258390, "APPID": APPID, "units": APPUNITS}
+request_parameters = {"id": 5261457, "APPID": APPID, "units": APPUNITS}
 
 # dictionary of names to objects
 objects = {}
@@ -220,6 +222,14 @@ def update_weather_data():
         if k in objects:
             objects[k]._set_value(v)
 
+@bacpypes_debugging
+class ReadPropertyMultipleApplication(
+   BIPSimpleApplication,
+   ReadWritePropertyMultipleServices,
+   DeviceCommunicationControlServices,
+   ):
+   pass
+
 
 @bacpypes_debugging
 def main():
@@ -239,7 +249,7 @@ def main():
         _log.debug("    - this_device: %r", this_device)
 
     # make a sample application
-    this_application = BIPSimpleApplication(this_device, args.ini.address)
+    this_application = ReadPropertyMultipleApplication(this_device, args.ini.address)
 
     # create the objects and add them to the application
     create_objects(this_application)

```

Otherwise the code I use is exactly what was above in the first comment - thus far I've only used the Python REPL with BAC0 - since BAC0 had the web interface and I just wanted to see a trend plot I haven't tried using a notebook yet. 

If you think running BAC0 in 'lite' mode with InfluxDB would work, I can try that. If the Bokeh parts don't have a future, no sense in trying to debug them!

ChristianTremblay, 2022-02-02:
[face palm]

no need to define `.chart()`

once the device is defined, just open the web interface and open the trend page

[image]

click here and select the point you want to trend.

Can you believe I wrote that... and didn't remember that trick... [sigh]

Anyway... regarding influx and Bokeh they are really solving different issues. This Bokeh stuff, the way it's done, it's for temporary trending. Everything is "in-memory"...
But influxDB... data is pushed to a DB that stores everything... then you can use Influx dashboard itself or Grafana... and seriously, BAC0 is a side project... I just can't compete with Grafana.

epaulson, 2022-02-03:
OK, cool. I saw it working! The Bokeh server /trends page is a little fidgety - sometimes when you load the page the Bokeh JS blows up because it's getting strange results from the server, and in other cases when the page loads some of the Bokeh JS starts but the point selection box never appears. But, in those cases, you just press reload and eventually you get a run where everything works - that's good enough for my use of just showing a few minutes of a trend once. 

If you're interested in digging more into the Bokeh thing, here's a browser screenshot. But again, I completely understand that BAC0 is a side project and don't at all feel bad about saying "won't fix" for this!

[image]

closed 2022-06-13 by github-actions[bot]