ChristianTremblay/BAC0 #319: Not getting devices after bacnet.discover

pisuke, 2022-03-18:
Hi, 

I'm trying to use the code in the `develop` branch and `master` branch.
In both cases, when I try the following code:

```python
import BAC0

bacnet = BAC0.connect() 
bacnet.discover()

print(bacnet.devices)
for bacnet_device in bacnet.devices:
    print(bacnet_device)
```

I get an empty Pandas dataframe in the bacnet.devices object.

This is the output of my script:

```bash
2022-03-18 15:34:29,947 - INFO    | Starting BAC0 version 22.01.06dev (Complete)
2022-03-18 15:34:29,947 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-03-18 15:34:29,947 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-03-18 15:34:29,947 - INFO    | Starting TaskManager
2022-03-18 15:34:29,949 - INFO    | Using ip : 192.168.1.148
2022-03-18 15:34:29,961 - INFO    | Starting app...
2022-03-18 15:34:29,961 - INFO    | BAC0 started
2022-03-18 15:34:29,961 - INFO    | Registered as Simple BACnet/IP App
2022-03-18 15:34:29,964 - INFO    | Update Local COV Task started
2022-03-18 15:34:31,114 - INFO    | Server started : http://192.168.1.148:8111
 * Serving Flask app 'BAC0.web.FlaskServer' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses.
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://192.168.1.148:8111/ (Press CTRL+C to quit)
2022-03-18 15:34:35,120 - INFO    | Found those networks : set()
2022-03-18 15:34:35,121 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
2022-03-18 15:34:42,126 - INFO    | Found those networks : set()
2022-03-18 15:34:42,127 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
Empty DataFrame
Columns: [Manufacturer, Address,  Device ID]
Index: []
Manufacturer
Address
 Device ID
```

A Wireshark capture show this:

```
2945	4.231313538	0.012400963	192.168.1.148	192.168.1.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,3057154 
2946	4.231426334	0.000112796	192.168.1.148	192.168.1.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,3057154 
3572	5.363605458	0.004860897	192.168.1.148	192.168.1.255	BACnet-NPDU	60	What-Is-Networknumber
3573	5.364709541	0.001104083	192.168.1.173	192.168.1.255	BACnet-NPDU	60	Networknumber-Is
5174	8.588284120	0.000449394	DeltaCon_10:38:7b	Broadcast	BACnet-APDU	60	Unconfirmed-REQ who-Is 3057154 3057154 
5175	8.588568938	0.000284818	192.168.1.173	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ who-Is 3057154 3057154 
6841	11.095690918	0.001657300	192.168.1.148	192.168.1.255	BACnet-APDU	60	Unconfirmed-REQ who-Is 0 4194303 
6842	11.096818978	0.001128060	192.168.1.173	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,4114459 
6843	11.097114914	0.000295936	192.168.1.127	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,1 
6844	11.101085151	0.003970237	192.168.1.141	192.168.1.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,1141 
7798	14.473124384	0.000048215	192.168.1.148	192.168.1.255	BACnet-NPDU	60	Who-Is-Router-To-Network
8023	15.603997890	0.005779054	DeltaCon_10:38:7b	Broadcast	BACnet-APDU	60	Unconfirmed-REQ who-Is 3057154 3057154 
8024	15.604290081	0.000292191	192.168.1.173	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ who-Is 3057154 3057154 
8352	16.464542572	0.013012814	192.168.1.148	192.168.1.255	BACnet-APDU	60	Unconfirmed-REQ who-Is 0 4194303 
8353	16.465918531	0.001375959	192.168.1.127	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,1 
8354	16.466363238	0.000444707	192.168.1.173	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ i-Am device,4114459 
8355	16.470363488	0.004000250	192.168.1.141	192.168.1.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,1141 
10075	22.614176006	0.000002617	DeltaCon_10:38:7b	Broadcast	BACnet-APDU	60	Unconfirmed-REQ who-Is 3057154 3057154 
10076	22.614337743	0.000161737	192.168.1.173	192.168.1.255	BACnet-APDU	62	Unconfirmed-REQ who-Is 3057154 3057154 
```

So I can see that devices in the network are responding.
What could be the issue preventing devices to be captured by BAC0?
Or am I using the library wrongly?

pisuke, 2022-03-27:
I've checked the same code on a windows machine and it finds the devices after a discover, while it doesn't on my mac and linux boxes. Is there anyone using BAC0 on linux or mac who has experienced a similar behaviour?

ChristianTremblay, 2022-04-14:
Can you try specify the subnet when you define your bacnet network ? (ex. ip/24)
I know there are some little nuances between Windows & Mac & Linux and BACnet discovery is highly tied to the subnet and broadcast address.

gnmerritt, 2022-04-28:
Another thing you might try is running a `bacnet.whois()` after your discovery call. We've found that the separate code path sometimes picks up devices that `bacnet.discover()` misses

closed 2022-07-06 by github-actions[bot]