ChristianTremblay/BAC0 #323: Device Disconnected when shouldn't be

jetran01, 2022-04-25:
Hi i have been able to perform "queries" to bacnet devices using vpn's and avoiding using BMMD's. but from  the 3 devices i can only reach 2, all 3 are on the same network and i have been granted permissions to access all 3 devices using the vpn.

The device that doesn't respond is a Sauter Device EY and although i can see it using yabe and nmap

[image]

[image]

I'm guessing it says filtered in nmap because i cannot reach it trough ping acknowledgement.

¿Is this required for BAC0 to work? (i got this clue with BAC0.lite() documentation)

Besides it is kind of different in comparison since it has another device within + the objects from the sensors connected to it. At the moment i have been trying to reach to the device just using its IP address mainly because i don't know how to reach it with MSTP # or the MAC # as u usually show on your examples.  

another interesting thing is that seems like every day at 7am it push information into other devices or at least i think it does.  can i intercept it with bac0 or it is to crazy?

By the way i cannot use WhoIs just unicast requests but this isn't an issue just context.

Hope you can help me

ChristianTremblay, 2022-05-05:
BACnet communication require UDP 47808, nothing else.

I suspect the devices you are looking for are on another network. 
Reaching an MSTP device is done by using `network:mac` syntax as address.

Intercept a message... not sure. You could probably see the packet in some way (wireshark for example)... But I didn't implement any packet sniffer in BAC0

jetran01, 2022-05-09:
Regarding the "another network" posible solution.
I  have done some "trace route" with my virtual bacnet devices in remote to the 3 devices and between one of them and the other 2 in local and they don't pass trough another router so they seem to be on the same network.
although i lost track of the remote "trace route" in the one that i can not reach (in local i can reach it of course)

closed 2022-07-16 by github-actions[bot]