ChristianTremblay/BAC0 #325: AWS linux BAC0 openvpn client

jetran01, 2022-05-09:
Hi @ChristianTremblay 

Have you ever encounter error 99 (cannot reach network) when trying to create a bacnet network on a tun0 adapter (openvpn client IP) in ubuntu 20.04?

I can just replicate this issue in an AWS instance with ubuntu 20.04 installed. In W10 in my local PC works perfectly fine with openvpn client.

Any hint?
BR

jetran01, 2022-05-09:
Hi @ChristianTremblay 
Seems like the problem isn't AWS but just ubuntu 20.04 since i could reproduce the error in a VM with the same OS.

BR

jetran01, 2022-05-18:
Hi @ChristianTremblay 

There is some kind of incompatibility between openvpn 2.5 on ubuntu 20.04 and BAC0 (laste version). (It works fine on W10)

Does this has been reported? Do you have any similar cases?

jetran01, 2022-05-18:
Error detected
>>python3
Python 3.6.9 (default, Mar 15 2022, 13:55:28) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>import BAC0
>>bacnet=BAC0.lite('10.8.0.6')
2022-05-18 13:32:17,087 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-05-18 13:32:17,087 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-05-18 13:32:17,087 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-05-18 13:32:17,087 - INFO    | Starting TaskManager
2022-05-18 13:32:17,089 - INFO    | Using ip : 10.8.0.6
2022-05-18 13:32:17,107 - ERROR   | an error has occurred: [Errno 99] Cannot assign requested address
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/BAC0/scripts/Base.py", line 235, in startApp
    subscription_contexts=self.subscription_contexts,
  File "/usr/local/lib/python3.6/dist-packages/BAC0/core/app/ScriptApplication.py", line 217, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/usr/local/lib/python3.6/dist-packages/bacpypes/bvllservice.py", line 102, in __init__
    self.broadcastPort = UDPDirector(self.addrBroadcastTuple, reuse=True)
  File "/usr/local/lib/python3.6/dist-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python3.6/asyncore.py", line 329, in bind
    return self.socket.bind(addr)
OSError: [Errno 99] Cannot assign requested address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/BAC0/scripts/Base.py", line 177, in __init__
    self.startApp()
  File "/usr/local/lib/python3.6/dist-packages/BAC0/scripts/Base.py", line 250, in startApp
    raise InitializationError("Error starting app: {}".format(error))
BAC0.core.io.IOExceptions.InitializationError: Error starting app: [Errno 99] Cannot assign requested address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/BAC0/scripts/Lite.py", line 158, in __init__
    **params
  File "/usr/local/lib/python3.6/dist-packages/BAC0/scripts/Base.py", line 180, in __init__
    "Gros probleme : {}. Address requested : {}".format(error, localIPAddr)
BAC0.core.io.IOExceptions.InitializationError: Gros probleme : Error starting app: [Errno 99] Cannot assign requested address. Address requested : 10.8.0.6

ChristianTremblay, 2022-05-18:
This is new material for me. I work a lot with OpenVPN but in Windows and it works all the time.

SteffenBraeunig, 2022-07-15:
Hi @ChristianTremblay

In a docker environment (host: ubuntu:20.04) I have exactly the same problem. In a local development environment (w10) it works fine. Example: client = BAC0.lite(). If I put the app into a docker image (base is python:3.10.5-slim), I get the following output:

2022-07-15 11:48:02,751 - INFO - Lite:__init__(151) - Using ip : 172.17.0.2
2022-07-15 11:48:02,751 - DEBUG - Base:__init__(116) - Configurating app
2022-07-15 11:48:02,768 - DEBUG - Base:__init__(121) - To be able to run the web server, you must install pandas, bokeh, flask and flask_bootstrap
2022-07-15 11:48:02,768 - DEBUG - Base:__init__(124) - Those are not all installed so BAC0 will work in Lite mode only.
2022-07-15 11:48:02,768 - DEBUG - Base:startApp(188) - Create Local Device
2022-07-15 11:48:02,769 - DEBUG - Discover:_iam_request(248) -   - request: <bacpypes.apdu.IAmRequest(0) instance at 0x7fa74ab2bf40>
2022-07-15 11:48:02,770 - ERROR - Base:startApp(249) - an error has occurred: [Errno 99] Cannot assign requested address
2022-07-15 11:48:02,770 - DEBUG - Base:startApp(252) - finally
2022-07-15 11:48:02,770 - ERROR - app:handle_exception(32) - Gros probleme : Error starting app: [Errno 99] Cannot assign requested address. Address requested : 172.17.0.2
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/BAC0/scripts/Base.py", line 231, in startApp
    self.this_application = BAC0Application(
  File "/usr/local/lib/python3.10/site-packages/BAC0/core/app/ScriptApplication.py", line 217, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/usr/local/lib/python3.10/site-packages/bacpypes/bvllservice.py", line 102, in __init__
    self.broadcastPort = UDPDirector(self.addrBroadcastTuple, reuse=True)
  File "/usr/local/lib/python3.10/site-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/local/lib/python3.10/asyncore.py", line 333, in bind
    return self.socket.bind(addr)
OSError: [Errno 99] Cannot assign requested address

jetran01, 2022-07-15:
this adress "172.17.0.2" match the address of your docker or the one your vpn is giving you?

SteffenBraeunig, 2022-07-15:
@jetran01 

"172.17.0.2" is the dynamic lokal ip address of the running docker container. Maybe it helps you with the vpn problem.

    def __get_ip(self):
        s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
        s.connect(("8.8.8.8", 80))
        ip, port = s.getsockname()
        return ip

jetran01, 2022-07-15:
and what ip does the vpn assign to you?

closed 2022-09-21 by github-actions[bot]