ChristianTremblay/BAC0 #326: Error While reading from a device

ninjas005, 2022-05-13:
I am getting this error randomly while `read_multiple` data from a device.

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/iot_gateway/Bacnet/bacnet.py", line 98, in run
    field['value'] = 1 if device.read_property(property) == 'active' else 0
  File "/home/pi/.local/lib/python3.9/site-packages/BAC0/core/devices/Device.py", line 734, in read_property
    val = self.properties.network.read(
  File "/home/pi/.local/lib/python3.9/site-packages/BAC0/core/io/Read.py", line 218, in read
Connection Active.
    raise NoResponseFromController(
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : Timeout
```

I am not getting the reason for this Timeout. I am getting continuous ping from the controller without interruption.

ChristianTremblay, 2022-05-18:
Is this a really big network ? A lot of devices ?

ninjas005, 2022-05-22:
No just 3 devices. With about 95 properties in total

closed 2022-07-29 by github-actions[bot]