ChristianTremblay/BAC0 #329: Adjust COV on local objects?

deve87, 2022-05-26:
Hi.

How can I adjust COV in local points?
It's sending COV even though there is no changes in the object

bacnet["object"].covIncrement = 0.1 don't seem to work?

2022-05-26 21:35:56,963 - INFO    | Sending COV for object
2022-05-26 21:36:02,066 - INFO    | Sending COV for object

deve87, 2022-05-27:
If I comment out:
deferred(cov_detection.send_cov_notifications)
in UpdateCOV.py

Local COV seems to work with the covIncrement value.

COVIncrementCriteria in bacpypes cov.py compare the old_value, new_value with the covIncrement and gives out True/False and then sends COVnotifications.

I'm not good in python and use it only for simple tasks. Can somebody please explain what the Update_local_COV in UpdateCOV.py is for?

ChristianTremblay, 2022-06-14:
I'll have to re-dive in it. I have been on other things for a long time. This is related to bacpypes and the way BAC0 support that. I know bacpypes evolved also so ... thanks for your patience.

closed 2022-11-11 by github-actions[bot]