ChristianTremblay/BAC0 #331: I-Am response is not handled

JJK96, 2022-06-14:
Similar to #199, I run `bacnet.discover` and a device replies with I-Am, but the device is not shown in `bacnet.devices`.

My code:

```
import BAC0

import argparse
parser = argparse.ArgumentParser()
parser.add_argument("ip")
parser.add_argument("--port", "-p", default=47809)
args = parser.parse_args()

bbmdIP = '{}:{}'.format(args.ip, args.port)
bbmdTTL = 30
#bacnet = BAC0.connect()
bacnet = BAC0.lite(ip="10.50.172.8/27", bbmdAddress=bbmdIP, bbmdTTL=bbmdTTL)
bacnet.discover()
bacnet.whois()
```

My run:
```
$ python -i bac0test.py 10.50.172.1
2022-06-14 08:33:08,611 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-06-14 08:33:08,611 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-06-14 08:33:08,611 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-06-14 08:33:08,611 - INFO    | Starting TaskManager
2022-06-14 08:33:08,611 - INFO    | Using ip : 10.50.172.8
2022-06-14 08:33:08,619 - INFO    | Starting app...
2022-06-14 08:33:08,621 - INFO    | BAC0 started
2022-06-14 08:33:08,621 - INFO    | Registered as Foreign Device
2022-06-14 08:33:08,622 - INFO    | Update Local COV Task started
2022-06-14 08:33:12,626 - INFO    | Found those networks : set()
2022-06-14 08:33:12,626 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
>>> bacnet.devices
[]
```

The wireshark capture:
```
No.	Time	Source	Destination	Protocol	Length	Info
3	1.186283436	10.50.172.8	10.50.172.1	BVLC	48	BACnet Virtual Link Control BVLC Function Register-Foreign-Device 
4	1.191990794	10.50.172.1	10.50.172.8	BVLC	60	BACnet Virtual Link Control BVLC Function BVLC-Result 
5	2.047299096	fa:22:ee:95:db:e1	Broadcast	ARP	60	Who has 10.50.172.2? Tell 10.50.172.1
6	3.190591593	10.50.172.8	10.50.172.1	BACnet-NPDU	49	Who-Is-Router-To-Network
7	3.192509163	10.50.172.1	10.50.172.31	BACnet-NPDU	60	Who-Is-Router-To-Network
8	4.647965415	fa:22:ee:95:db:e1	Broadcast	ARP	60	Who has 10.50.172.2? Tell 10.50.172.1
9	5.193648028	10.50.172.8	10.50.172.1	BACnet-APDU	56	Unconfirmed-REQ who-Is 0 4194303 
10	5.200744991	10.50.172.1	10.50.172.31	BACnet-APDU	62	Unconfirmed-REQ who-Is 0 4194303 
11	5.200745435	10.50.172.1	10.50.172.8	BACnet-APDU	63	Unconfirmed-REQ i-Am device,10159 
12	5.663388031	fa:22:ee:95:db:e1	Broadcast	ARP	60	Who has 10.50.172.2? Tell 10.50.172.1
13	6.687427657	fa:22:ee:95:db:e1	Broadcast	ARP	60	Who has 10.50.172.2? Tell 10.50.172.1
14	8.198562819	10.50.172.8	10.50.172.1	BACnet-APDU	50	Unconfirmed-REQ who-Is 
15	8.204809835	10.50.172.1	10.50.172.31	BACnet-APDU	60	Unconfirmed-REQ who-Is 
16	8.204810149	10.50.172.1	10.50.172.8	BACnet-APDU	63	Unconfirmed-REQ i-Am device,10159 
```

You can see that for both the discover, and the whois, an i-Am response is received but not processed.

LytixDev, 2022-06-17:
I have the samme issue, only I get
`"ValueError: invalid literal for int() with base 10: '10.205.34.104'"`
after running `bacnet.discover()`.

When looking at wireshark or other tools there are clearly devices on the network that are not picked up.

LytixDev, 2022-06-20:
The issue in my case was fixed when removing the default gateway in the TCP/IPv4 properties.

JJK96, 2022-06-20:
That is in system settings? Anyway, that is not a fix, it's a workaround

ChristianTremblay, 2022-06-20:
> The issue in my case was fixed when removing the default gateway in the TCP/IPv4 properties.

The format to specify IP address in BAC0 is "ip/mask"
You don't specify the gateway.

The mask will serve calculate the broadcast address used to send whois and receive i-am mesages (and other broadcast messsages)

ChristianTremblay, 2022-06-20:
> Similar to #199, I run `bacnet.discover` and a device replies with I-Am, but the device is not shown in `bacnet.devices`.
>
> My code:
>
> ```
> import BAC0
>
> import argparse
> parser = [...]

If you run the same code in the REPL (not via a `.py` file. Do you get the same result ?

I suspect the script just ends before receiving responses.

ChristianTremblay, 2022-06-20:
argg.... nevermind...just saw (late) you use -i

ChristianTremblay, 2022-06-20:
One possibility is related to port. By default, BAC0 uses 47808. Your BBMD looks like it's using 47809.
Technically, for direct connection, it won't matter. But for discovery features using the broadcast, there can be an issue.

What if you try

```
import argparse
parser = argparse.ArgumentParser()
parser.add_argument("ip")
parser.add_argument("--port", "-p", default=47809)
args = parser.parse_args()

bbmdIP = '{}:{}'.format(args.ip, args.port)
bbmdTTL = 30
#bacnet = BAC0.connect()
bacnet = BAC0.lite(ip="10.50.172.8:47809/27", bbmdAddress=bbmdIP, bbmdTTL=bbmdTTL)
bacnet.discover()
bacnet.whois()

```

JJK96, 2022-06-20:
I no longer have the device, so I do not have the opportunity to test anymore... thanks for your suggestions though!