ChristianTremblay/BAC0 #333: problems with writeFunction on binaries Inputs

Sekiro-kost, 2022-06-22:
Hello Christian

I have a problem with writing values to binary inputs

In my code, I have this : 

```
   value = 'inactive'

  write_vals = f'{eqnetwork} {object_type} {object_instance} presentValue {value} - {priority}'         
 bacnet.write(write_vals)
```

I have an invalid literal for int() with base 10 'inactive' error

However, if I try to put 0 instead of inactive, I get an outOfRange error, and if I try to put false, I also get a base 10 error

However, I thought it worked with active or inactive values on binaries. 

Do you have any idea of the problem here?

ChristianTremblay, 2022-06-22:
Normally, Binary Inputs are read only. Writable if `out_of_service` is true.

Sekiro-kost, 2022-06-22:
Thx for your reply. 

What do you mean by : Writable if out_of_service is true ?

ChristianTremblay, 2022-06-22:
BACnet objects can have a property (like present_value is a property of an object). 

If someone write true to the property `out_of_service`, if implemented, it will disconnect the `binaryInput` from the hardware and you can then modify the presentValue by writing to it. 

But it is not a common way to operate and it is meant for troubleshooting and testing cases.

Sekiro-kost, 2022-06-22:
Hmm oki, thanks

Last question: for MultiStateValue, what is the property to access the different values available?

Is it statusFlags?

Sincerely

ChristianTremblay, 2022-06-23:
in BACnet it's the `state_text` property
in a BAC-.device, look for device.properties.units_state

closed 2022-08-30 by github-actions[bot]