ChristianTremblay/BAC0 #336: 'WritePropertyMultipleError' object has no attribute 'errorCode'

JerryXinhui, 2022-07-18:
When I try to use bacnet.writeMultiple to write a not exist object of the device, the error message is 'WritePropertyMultipleError' object has no attribute 'errorCode'.  But in the wireshark I can see the clear reason "unknown object".  How can I get the real reason. For sometimes,  I need the real reason to do some judgment.
And if I use bacnet.write to write a not exist object of the device,  I can get the real reason. 
[image]

bbartling, 2022-07-18:
I'm not someone to answer the question, but what are the use cases for `writeMultiple` versus just `write`? Would you have an example where this is used?

JerryXinhui, 2022-07-19:
I‘m not sure I understand your question correctly. But sometimes,  we use writeMultiple to write multiple object to test the device works well under this condition. And also , we use writeMultiple to write a non-existent object to the device in order  to test whether the device works well.

ChristianTremblay, 2022-07-19:
I don't have a lot of experience with writeMultiple as few devices I use supports it. 

I suspect that we will need to go back to the bacpypes internals to see the APdU error. I may have done something wrong in the implementation.

JoelBender, 2022-07-20:
In the ASN.1 of the standard, the `WritePropertyMultiple-Error` is described like this:
```
WritePropertyMultiple-Error ::= SEQUENCE {
    error-type [0] Error,
    first-failed-write-attempt [1] BACnetObjectPropertyReference
}
```
Which in BACpypes is translated to this:
```python
class WritePropertyMultipleError(ErrorSequence):
    sequenceElements = \
        [ Element('errorType', ErrorType, 0)
        , Element('firstFailedWriteAttempt', ObjectPropertyReference, 1)
        ]
```
The first element of the sequence will be a context encoded `ErrorType` structure, which itself contains the `errorClass` and `errorCode`.  The second element, also context encoded, is the reference to the first failed object and property (and possibly array index) that failed.

The weirdness is in **Clause 21.4** where most Result(-) errors are just errors, but some provide additional information and "include" an error.

Chris-Wang-bjc, 2022-07-28:
@JoelBender I have same problem, so how to solve it? Can I define error code and error class by myself? I am looking forward to your reply.

JoelBender, 2022-07-28:
If you call `apdu.debug_contents()` you'll see all of the content that has been decoded.  Check to make sure `hasattr(apdu, 'errorType')` and then `apdu.errorType.errorClass` and `apdu.errorType.errorCode` will have the rest of what you are looking for.

Chris-Wang-bjc, 2022-07-28:
Thanks for your reply, but sorry,  I‘m not sure I understand your reply correctly. You can tell me more details about "hasattr(apdu, 'errorType')"? I hope the error message will return the real reason instead of "no attribute" error, thanks.