ChristianTremblay/BAC0 #338: network number is 0

Ferkelcode, 2022-07-19:
Hi,

I used 
bacnet = BAC0.connect(ip='192.168.155.200/0')

then it shows 
2022-07-19 16:27:24,825 - INFO    | 192.168.155.2 network number is 0
2022-07-19 16:27:27,810 - INFO    | Found those networks : {0}
2022-07-19 16:27:27,810 - INFO    | Discovering network 0
2022-07-19 16:27:34,106 - INFO    | 192.168.155.2 network number is 0
2022-07-19 16:27:37,986 - INFO    | Found those networks : {0}
2022-07-19 16:27:37,986 - INFO    | Discovering network 0

what is the reason?

Thank you!

bbartling, 2022-07-19:
Are you trying to connect to a specific device? If so try this below as an example to look at but you need to know the device ID:

```
import BAC0,time

bacnet = BAC0.lite()
time.sleep(2)

rtu = BAC0.device('192.168.155.200', 1100, bacnet)

print(rtu.points)

bacnet.disconnect()
```

Else is `192.168.155.200` BACnet router and you are trying to see what is on BACnet MSTP network `0`?

Sometimes I add this to my code to see what I can see:

```
devices = bacnet.whois(global_broadcast=True)
device_mapping = {}
for device in devices:
    if isinstance(device, tuple):
        device_mapping[device[1]] = device[0]
        print("Detected device %s with address %s" % (str(device[1]), str(device[0])))
print(device_mapping)
print((str(len(device_mapping)) + " devices discovered on network."))

```

ChristianTremblay, 2022-07-19:
`bacnet = BAC0.connect(ip='192.168.155.200/0')`
Means create a bacnet network representation iun my machine using my local IP 192.168.155.200 with subnet 0

I suspect this is not what you meant

Ferkelcode, 2022-07-20:
Thank you very much for the answer.

I want to connect to the network 192.168.155.200, and to see what device is in the network. I actually don't know the subnet, what can I do?

bbartling, 2022-07-20:
I would just guess it either one of these 3: `255.0.0.0` or `255.255.0.0` or `255.255.255.0` or i think represented like this if I understand the question correctly:
`ip='192.168.155.200/8'` or
`ip='192.168.155.200/16'` or
`ip='192.168.155.200/24'`

Try and see if one works?

Ferkelcode, 2022-07-21:
Thanks so much.

I tried ip='192.168.155.200/8' ,   ip='192.168.155.200/16' and ip='192.168.155.200/9', both showed:

> 
> 2022-07-21 10:01:20,950 - INFO    | Found those networks : set()
> 2022-07-21 10:01:20,950 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
> 2022-07-21 10:01:31,135 - INFO    | Found those networks : set()
> 2022-07-21 10:01:31,135 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
> 
> Process finished with exit code 0

And for ip='192.168.155.200/24' , it showed:

> 2022-07-21 10:02:09,602 - INFO    | Found those networks : {0}
> 2022-07-21 10:02:09,602 - INFO    | Discovering network 0
> 2022-07-21 10:02:16,023 - INFO    | 192.168.155.2 network number is 0
> 2022-07-21 10:02:19,825 - INFO    | Found those networks : {0}
> 2022-07-21 10:02:19,825 - INFO    | Discovering network 0
> 
> Process finished with exit code 0

Ferkelcode, 2022-07-21:
Thank you for your time.

I tried:

> import BAC0
> 
> bacnet = BAC0.connect(ip='192.168.155.200/24')
> bacnet.discover()
> 
> print(bacnet.devices)

There in the table I saw the device, it should be the correct one. 

However the Device ID = 0, and the browser also shows the Number of devices is 0

Ferkelcode, 2022-07-21:
But with 

import BAC0

> 
> bacnet = BAC0.connect(ip='192.168.155.200/24')
> my_device = BAC0.device('192.168.155.200', 0, bacnet)
The INFO became:

> 2022-07-21 12:48:04,870 - INFO    | 192.168.155.2 network number is 0
> 2022-07-21 12:48:07,438 - INFO    | Found those networks : {0}
> 2022-07-21 12:48:07,438 - INFO    | Discovering network 0
> 2022-07-21 12:48:10,472 - INFO    | **Changing device state to DeviceDisconnected'>**
> 2022-07-21 12:48:20,492 - WARNING | Error connecting: APDU Abort Reason : Timeout
> 2022-07-21 12:48:20,492 - WARNING | Offline: provide database name to load stored data.
> 2022-07-21 12:48:20,493 - WARNING | Ex. controller.connect(db = 'backup')

when I print(my_device), it is:  / Disconnected

Can I change it somehow?

bbartling, 2022-07-21:
What does this do you for you?

```
import BAC0,time

bacnet = BAC0.lite()
time.sleep(2)

my_device = BAC0.device('192.168.155.200', 0, bacnet)

print(my_device.points)

bacnet.disconnect()
```

Maybe try changing your BACnet device ID and network to something other than 0. That seems odd, I have never seen that in the real world. Can you change your device ID 5 and your network something like different like 9 just for troubleshooting purposes?

bbartling, 2022-07-21:
What is your goal? To log data or make reads and write commands?

Ferkelcode, 2022-07-21:
Thank you.

I used device ID 0, because `bacnet.whois` showed 2 devices:  [('192.168.155.1', 930614), ('192.168.155.2', 0)]
I want to look into the second one  ('192.168.155.2', 0). Because from `bacnet.devices` the Manufacturer seems right, so I think this is the device I am looking for?

>                         Manufacturer        Address   Device ID
> Name                                                           
> BAC0                   SERVISYS inc.  192.168.155.1      930614
> Beckhoff_0  Beckhoff Automation GmbH  192.168.155.2           0

My goal is to make read and write commands.

bbartling, 2022-07-21:
I mostly use BAC0 to do demand response algorithm testing which is alot of reading, writing, releasing HVAC system points.

And never use BAC0.connect or BAC0.device but maybe in the future if do more data logging.

See these as examples of reading and writing and releasing:
- https://github.com/bbartling/Load-Shredder/tree/develop/Load-Shed
- https://gist.github.com/bbartling/88e95557b0b297bea721c0f285657d53

Ferkelcode, 2022-07-21:
Thank you so much!!!

ChristianTremblay, 2022-08-18:
When declaring your BACnet network, be sure to use the local IP in the same subnet than 

> Thank you.
> 
> I used device ID 0, because `bacnet.whois` showed 2 devices: [('192.168.155.1', 930614), ('192.168.155.2', 0)] I want to look into the second one ('192.168.155.2', 0). Because from `bacnet.devices` the Manufacturer seems right, so I think this is the device I am looking for?
> 
> [image]
> 

```
dev = BAC0.device('192.168.155.2', 0, bacnet)
```

should work.... why do you use `.200`?

device = address of the BACnet device
connect or lite -> local IP of the thing that run BAC0, on the subnet of other BACnet devices

closed 2022-10-25 by github-actions[bot]