ChristianTremblay/BAC0 #340: How do I write a Bitstring Presentvalue?

Chris-Wang-bjc, 2022-08-04:
I can't write a Bitstring Presentvalue right now, the error info is: TypeError('invalid constructor datatype')

bbartling, 2022-08-04:
If you dont mind me asking, what is a `BitString` or what is your use case? Strings over BACnet?

So it looks like in bacpypes [objects.py](https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#:~:text=class%20bitstringvalueobject(object)%3A) there is such a thing.

Make sure your syntax is same as BACpypes...Maybe something like if I understand `bitstringValue`:

BAC0 read request:

` read_vals = f'{address} bitstringValue {object_instance} presentValue'`

Chris-Wang-bjc, 2022-08-05:
@bbartling Thanks for your reply. Sorry, I may not have explained my problem, it's write property service, not read property service.
There is no problem with my code syntax, I can write the presentValue of other data types correctly, but the presentValue of "BitString" data type, when I use the same syntax to write a presentValue of this data type, it throws "TypeError('invalid constructor datatype')", so, can you please try to write a presentValue whose data type is BitString? Thanks!

bbartling, 2022-08-05:
Can you show your code? I can see if I can get it to work on my end.

bbartling, 2022-08-05:
Another thing you may want to do is find out if your device supports BitString data type.

Chris-Wang-bjc, 2022-08-08:
[image]
Here is my code. I'm not sure the format of the value that I write correctly. I'm sure my device supports BitString data type, because I'm trying write it by VTS, it's successful. So please help me check it on your device, thanks!

bbartling, 2022-08-08:
try getting rid of the curly brackets? `{}` ...I personally have never heard of bitString before with BACnet. What is your application again? HVAC in the field or security system? Or are you a BACnet developer on a test bench testing your real device with BAC0?

ChristianTremblay, 2022-08-18:
BAC0 does not support Bitstring actually.
To make it work, you could get inspiration from https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/functions/Text.py

and build the request.

Bottom line is : 

- you create a complete request that will be sent to IOCB with deferred
- in this request, the value will be "cast" to BitString in some way

A little trial and errors will probably be required.

I don't have any controller at hand support BitString....impossible for me to test for now.

Chris-Wang-bjc, 2022-08-19:
Thank you very much, I will take more try.

closed 2022-10-25 by github-actions[bot]