ChristianTremblay/BAC0 #346: tips for influx db

bbartling, 2022-09-12:
Would you have any tips for how to query for data in influx? Not sure what I am doing but I did [a tutorial ](https://www.arubacloud.com/tutorial/how-to-install-influxdb-on-ubuntu-20-04.aspx)for to install influx 1.8 on Ubuntu 20.04.....seems to be working??

No errors, ha. Whats also odd is I cannot get at the web interface for the database....if I go this machines url port 8086 I just get a 404 returned.

```
import BAC0
import time
import pandas as pd

bacnet = BAC0.lite()

obj_list = [('file', 1),
            ('analogInput', 2),
            ('analogInput', 3)]

my_device = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=obj_list,
                        segmentation_supported=False)

time.sleep(3)

column_names = []
for point in my_device.points:
    print(point.properties.name)
    column_names.append(point.properties.name)

df = pd.DataFrame(columns=column_names)

def appendDictToDF(df, dictToAppend):
    df = pd.concat([df, pd.DataFrame.from_records([dictToAppend])])
    df = df.fillna(method='ffill').fillna(method='bfill')
    return df

def update_app():

    while True:

        for point in my_device.points:

            data = {
                point.properties.name: point.value
            }

            df = appendDictToDF(df, data)

            print(df)

            time.sleep(60)

bacnet.disconnect()
print('disconnected')
```

Also notice influx or BAC0 created some files:

- Device_201201.bin
- Device_201201.db

Just curious to see if I can look at the database data before the Grafana steps in the readme. Thanks for any tips if you have time to respond...

bbartling, 2022-09-17:
Still cant figure out how to browse into the influx db....on my test bench if I go to a http://192.168.0.100:8086/

I just get a 404 returned

(running Ubuntu 20.04 on a spare machine)

```
$ influx -database 'metrics' -host '0.0.0.0' -port '8086'
Connected to http://0.0.0.0:8086 version 1.8.10
InfluxDB shell version: 1.8.10
> auth
username: admin
password:
> SHOW USERS
user  admin
----  -----
admin true

> CREATE DATABASE BENSHOUSE
> USE BENSHOUSE
Using database BENSHOUSE
> SHOW DATABASES
name: databases
name
----
_internal
BENSHOUSE
>
```

On my test bench I can create a database...but cant figure out how to get the `token` .

Does BAC0 create a database to and a table automatically? Or am I missing anything?

bbartling, 2022-09-17:
trying to upgrade to influx V2.....https://askubuntu.com/questions/1429908/install-influx-database-v2-on-ubuntu

ChristianTremblay, 2022-09-17:
You absolutely need v2
The python plugin is different in V2 and I don't plan any support for <2

ChristianTremblay, 2022-09-17:
https://bac0.readthedocs.io/en/develop/database.html?highlight=database

closed 2022-09-17 by ChristianTremblay