ChristianTremblay/BAC0 #348: BAC0.devices on multiple devices

bbartling, 2022-09-17:
Hi,

Is it possible to `connect` to multiple devices?

I have 2 old mstp devices on a test bench one called `furnace` and the other called `hotwaterheatr`:

```
import BAC0
import time

bacnet = BAC0.lite()

furnace_obj_list = [('file', 1),
            ('analogInput', 2)]

hotwaterhtr_obj_list = [('file', 1),
            ('analogInput', 2)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

hotwaterheatr = BAC0.device('12345:2',
                        201202,
                        bacnet,
                        object_list=hotwaterhtr_obj_list,
                        segmentation_supported=False)

time.sleep(3)

devices = [furnace,hotwaterheatr]

def update_app():

    while True:
        for device in devices:
            for point in device.points:

                data = {
                    point.properties.name: point.value
                }

                print(data)
                time.sleep(60)

bacnet.disconnect()
print('disconnected')
```
Thanks for any tips sort of curious to build this into connecting multiple devices to an influx db using the built in BAC0 method of `connect` vs doing a bunch read requests on the network....

If I run this I get a:

```
$ python3.8 test.py
2022-09-17 10:48:40,328 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-09-17 10:48:40,328 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-09-17 10:48:40,328 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-09-17 10:48:40,329 - INFO    | Starting TaskManager
2022-09-17 10:48:40,348 - INFO    | Using ip : 192.168.0.100
2022-09-17 10:48:40,382 - INFO    | Starting app...
2022-09-17 10:48:40,383 - INFO    | BAC0 started
2022-09-17 10:48:40,383 - INFO    | Registered as Simple BACnet/IP App
2022-09-17 10:48:40,385 - INFO    | Update Local COV Task started
2022-09-17 10:48:40,386 - INFO    | Changing device state to DeviceDisconnected'>
2022-09-17 10:48:40,430 - INFO    | 192.168.0.190 router to [12345]
2022-09-17 10:48:40,661 - INFO    | Changing device state to RPDeviceConnected'>
2022-09-17 10:48:41,012 - INFO    | Device 201201:[TEST1] found... building points list
2022-09-17 10:48:41,460 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2022-09-17 10:48:41,462 - INFO    | Ready!
2022-09-17 10:48:41,462 - INFO    | Device defined for normal polling with a delay of 10sec
2022-09-17 10:48:41,462 - INFO    | Polling started, values read every 10 seconds
2022-09-17 10:48:41,463 - INFO    | Changing device state to DeviceDisconnected'>
2022-09-17 10:48:41,580 - WARNING | Unknown object 12345:2 device 201202 objectName
Traceback (most recent call last):
  File "test.py", line 22, in <module>
    hotwaterheatr = BAC0.device('12345:2',
  File "/home/ben/.local/lib/python3.8/site-packages/BAC0/core/devices/Device.py", line 199, in __init__
    self.new_state(DeviceDisconnected)
  File "/home/ben/.local/lib/python3.8/site-packages/BAC0/core/devices/Device.py", line 215, in new_state
    self._init_state()
  File "/home/ben/.local/lib/python3.8/site-packages/BAC0/core/devices/Device.py", line 853, in _init_state
    self.connect()
  File "/home/ben/.local/lib/python3.8/site-packages/BAC0/core/devices/Device.py", line 865, in connect
    name = self.properties.network.read(
  File "/home/ben/.local/lib/python3.8/site-packages/BAC0/core/io/Read.py", line 209, in read
    raise UnknownObjectError("Unknown object {}".format(args))
BAC0.core.io.IOExceptions.UnknownObjectError: Unknown object 12345:2 device 201202 objectName

```

ChristianTremblay, 2022-09-17:
2 different devices with the same MAC address ?
12345:2 for both...

network:mac

bbartling, 2022-09-18:
Oh snap....typo thanks @ChristianTremblay at least the code does error out now, but why does it exit right away?

```
import BAC0
import time

bacnet = BAC0.lite()

furnace_obj_list = [('file', 1),
            ('analogInput', 2)]

hotwaterhtr_obj_list = [('file', 1),
            ('analogInput', 2)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

hotwaterheatr = BAC0.device('12345:6',
                        201202,
                        bacnet,
                        object_list=hotwaterhtr_obj_list,
                        segmentation_supported=False)

time.sleep(3)

devices = [furnace,hotwaterheatr]

def update_app():

    while True:
        for device in devices:
            for point in device.points:

                data = {
                    point.properties.name: point.value
                }

                print(data)
                time.sleep(60)

bacnet.disconnect()
print('disconnected')
```

Am I doing any funny in the code why it exits right away?
```
2022-09-18 07:52:45,978 - INFO    | Starting app...
2022-09-18 07:52:45,979 - INFO    | BAC0 started
2022-09-18 07:52:45,979 - INFO    | Registered as Simple BACnet/IP App
2022-09-18 07:52:45,982 - INFO    | Update Local COV Task started
2022-09-18 07:52:45,983 - INFO    | Changing device state to DeviceDisconnected'>
2022-09-18 07:52:46,021 - INFO    | 192.168.0.190 router to [12345]
2022-09-18 07:52:46,302 - INFO    | Changing device state to RPDeviceConnected'>
2022-09-18 07:52:46,724 - INFO    | Device 201201:[TEST1] found... building points list
2022-09-18 07:52:47,192 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2022-09-18 07:52:47,193 - INFO    | Ready!
2022-09-18 07:52:47,193 - INFO    | Device defined for normal polling with a delay of 10sec
2022-09-18 07:52:47,193 - INFO    | Polling started, values read every 10 seconds
2022-09-18 07:52:47,194 - INFO    | Changing device state to DeviceDisconnected'>
2022-09-18 07:52:47,406 - INFO    | Changing device state to RPDeviceConnected'>
2022-09-18 07:52:47,752 - INFO    | Device 201202:[TEST2] found... building points list
2022-09-18 07:52:48,361 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2022-09-18 07:52:48,362 - INFO    | Ready!
2022-09-18 07:52:48,363 - INFO    | Device defined for normal polling with a delay of 10sec
2022-09-18 07:52:48,363 - INFO    | Polling started, values read every 10 seconds
2022-09-18 07:52:51,366 - INFO    | Wait while stopping polling
2022-09-18 07:52:51,366 - INFO    | Polling stopped
2022-09-18 07:52:51,958 - INFO    | Device saved to Device_201201.db
2022-09-18 07:52:51,958 - INFO    | Changing device state to DeviceFromDB'>
2022-09-18 07:52:51,958 - INFO    | Initializing DB
2022-09-18 07:52:51,963 - INFO    | Device restored from db
2022-09-18 07:52:51,963 - INFO    | You can reconnect to network using : "device.connect(network=bacnet)"
2022-09-18 07:52:51,963 - INFO    | Wait while stopping polling
2022-09-18 07:52:51,963 - INFO    | Polling stopped
2022-09-18 07:52:52,239 - INFO    | Device saved to Device_201202.db
2022-09-18 07:52:52,239 - INFO    | Changing device state to DeviceFromDB'>
2022-09-18 07:52:52,239 - INFO    | Initializing DB
2022-09-18 07:52:52,243 - INFO    | Device restored from db
2022-09-18 07:52:52,243 - INFO    | You can reconnect to network using : "device.connect(network=bacnet)"
2022-09-18 07:52:52,243 - INFO    | Stopping all tasks
2022-09-18 07:52:52,247 - INFO    | Stopping TaskManager
2022-09-18 07:52:52,253 - INFO    | Ok all tasks stopped
2022-09-18 07:52:52,254 - INFO    | BACnet stopped
disconnected
```

I also notice these two files get created now that I have influx installed on this machine:
[image]

bbartling, 2022-09-18:
Oh snap a typo this works works now....is there anything funny with doing a `time.sleep` at the end of the while loop under the `print(data)`? Does that hose up all processes like the BACnet polling??? Just curious if I should do anything different their 

```
import BAC0
import time

bacnet = BAC0.lite()

furnace_obj_list = [('file', 1),
            ('analogInput', 2)]

hotwaterhtr_obj_list = [('file', 1),
            ('analogInput', 2)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

hotwaterheatr = BAC0.device('12345:6',
                        201202,
                        bacnet,
                        object_list=hotwaterhtr_obj_list,
                        segmentation_supported=False)

time.sleep(3)

devices = [furnace,hotwaterheatr]

while True:
    for device in devices:

        for point in device.points:

            data = {
                device.properties.name : point.value
            }

            print(data)
            time.sleep(60)

bacnet.disconnect()
print('disconnected')
```

bbartling, 2022-09-18:
Actually should I use `recurring task`? Any chance you could comment on this below? I moved everything to a main function....sort of scabbed what you wrote in #343 

```
import BAC0
from BAC0.tasks.RecurringTask import RecurringTask 
import time

bacnet = BAC0.lite()

furnace_obj_list = [('file', 1),
            ('analogInput', 2)]

hotwaterhtr_obj_list = [('file', 1),
            ('analogInput', 3)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

hotwaterheatr = BAC0.device('12345:6',
                        201202,
                        bacnet,
                        object_list=hotwaterhtr_obj_list,
                        segmentation_supported=False)

time.sleep(3)

devices = [furnace,hotwaterheatr]

def printData():
    for device in devices:
        for point in device.points:

            data = {
                device.properties.name : point.value
            }

            print(data)
            time.sleep(60)

def main():
    task1 = RecurringTask(printData,delay=60)
    task1.start()

    while True:
        time.sleep(0.01)

if __name__ == "__main__":
    main()

bacnet.disconnect()
print('disconnected')
```

ChristianTremblay, 2022-09-18:
``python
import BAC0
from BAC0.tasks.RecurringTask import RecurringTask 
import time
import sys

bacnet = BAC0.lite()

furnace_obj_list = [('analogInput', 2)]

hotwaterhtr_obj_list = [('analogInput', 3)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

hotwaterheatr = BAC0.device('12345:6',
                        201202,
                        bacnet,
                        object_list=hotwaterhtr_obj_list,
                        segmentation_supported=False)

time.sleep(3)

devices = [furnace,hotwaterheatr]

def printData():
    for device in devices:
        for point in device.points:
            data = {
                device.properties.name : point.lastValue
            }
            print(data)

def main():
    task1 = RecurringTask(printData,delay=60)
    task1.start()

    while True:
        time.sleep(0.01)

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        bacnet.disconnect()
        print('disconnected')
        sys.exit(0)
```

bbartling, 2022-09-19:
Thanks Christian.....I may bug you another issue how to build this into the `complete` version with the web browser charts of trends.

closed 2022-09-19 by bbartling