ChristianTremblay/BAC0 #349: Scanning BACnet devices from a BACnet controller

mohakbhalla7, 2022-09-17:
Hi @ChristianTremblay 
I'll try my best to explain my use case and the problem I am facing.

Use Case:
So I have a BACnet setup with me where there is a BACnet IP device connected with a BACnet/IP Controller, and that controller is connected with my RPi module through ethernet. Using BAC0, I want to write a script in my RPi to run a scan and get the details of those BACnet/IP devices connected with the BACnet controller.

Error:
I simply tried to connect to the network by using `BAC0.lite()` and this was the error that I got.
```
bacnet = BAC0.lite(ip='192.168.10.1/24')
2022-09-18 00:21:50,827 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-09-18 00:21:50,827 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-09-18 00:21:50,827 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-09-18 00:21:50,827 - INFO    | Using ip : 192.168.10.1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/bacnet_test/bacnet/lib/python3.7/site-packages/BAC0/scripts/Lite.py", line 158, in __init__
    **params
  File "/home/pi/bacnet_test/bacnet/lib/python3.7/site-packages/BAC0/scripts/Base.py", line 145, in __init__
    localIPAddr
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (192.168.10.1) invalid. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask')
```
The RPi has the `eth0` address as `192.168.10.51` and subnet-mask as `255.255.255.0`.
After getting this error I even tried defining multiple IP on `eth0` interface. But it still gave the same error.
```
eth0:1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  broadcast 192.168.10.255
        ether b8:27:eb:92:4a:31  txqueuelen 1000  (Ethernet)
```

After all this, I have a question. Is it possible to scan, read or write BACnet/IP devices connected with a BACnet controller using BAC0?

Any help is appreciated.
Thanks!

ChristianTremblay, 2022-09-17:
bacnet = BAC0.lite(ip='192.168.10.51/24')

mohakbhalla7, 2022-09-22:
So, I figured it out.
```
bacnet = BAC0.connect(ip='192.168.10.51/24', port=47809, bbmdAddress='192.168.10.51:47808', bbmdTTL=900)
```
I was able to connect using this command. What I did here is that, since `47808` port was not free to start `BAC0` application, I started `BAC0` on `47809` and since all the devices were discoverable on `192.168.10.51:47808`, I used it as the bbmdAddress. It was working fine, but this wasn't the ideal way.

I wasn't able to find which service was using `47808` port. After much thought, I figured it out and closed that service, making the `47808` port free. Then I was simply able to connect using
```
bacnet = BAC0.connect(ip='192.168.10.51/24')
```

closed 2022-09-22 by mohakbhalla7