ChristianTremblay/BAC0 #350: run complete version with bokeh server

bbartling, 2022-09-19:
Am I adding trends in correctly for the complete version?
https://bac0.readthedocs.io/en/latest/connect.html#complete

The dashboard is really nice looking....but no trends to plot:
[image]

No errors when I run this:

```
import BAC0
import time,sys

bacnet = BAC0.connect(ip='192.168.0.100/24')

furnace_obj_list = [('analogInput', 2),
                    ('analogInput', 3)]

furnace = BAC0.device('12345:2',
                        201201,
                        bacnet,
                        object_list=furnace_obj_list,
                        segmentation_supported=False)

time.sleep(5)

furnace['tempUoTwoBalco'].chart()
furnace['tempUoThreeBalco'].chart()

def main():

    while True:
        time.sleep(0.01)

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        bacnet.disconnect()
        print('disconnected')
        sys.exit(0)
```

ChristianTremblay, 2022-09-19:
Go in the trends page. And use the drop down took to add the trends you want.

bbartling, 2022-09-19:
Gosh not seeing a drop down....here?
[image]

ChristianTremblay, 2022-09-22:
When a device is registered, the objetcs should be visible in a dropdown

[image]

[image]

closed 2022-11-29 by github-actions[bot]