ChristianTremblay/BAC0 #352: InvalidTag: real application tag required

mohakbhalla7, 2022-09-22:
Hi @ChristianTremblay,

I am trying to read an `analogValue` object but I am getting this error:
```
>>> bacnet.read('50007:40 analogValue 932 presentValue')

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/BAC0/core/io/Read.py", line 167, in read
    value = apdu.propertyValue.cast_out(datatype)
  File "/home/pi/.local/lib/python3.7/site-packages/bacpypes/constructeddata.py", line 1346, in cast_out
    helper = klass(self.tagList[0])
  File "/home/pi/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 798, in __init__
    self.decode(arg)
  File "/home/pi/.local/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 814, in decode
    raise InvalidTag("real application tag required")
bacpypes.errors.InvalidTag: real application tag required
```

I have gone through the read() function documentation and the source code, but couldn't find any argument accepting tags; and neither do I understand how to use `BACpypes` here to handle this error.

Any help is appreciated.
Thanks!

ChristianTremblay, 2022-09-29:
Is this controller leggit ? The datatype in the response is incorrect

pisuke, 2025-01-07:
Hello @ChristianTremblay I'm experiencing a similar issue with a Tridium device that has issues with an analogValue.

`BAC0.core.io.IOExceptions.UnknownPropertyError: Unknown property 100.87.227.131 analogValue 10127 units`

By wiresharking we get:

```
Frame 241801: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface eno1, id 0
Ethernet II, Src: TexasIns_a1:c8:62 (34:08:e1:a1:c8:62), Dst: 64:4e:d7:50:00:63 (64:4e:d7:50:00:63)
Internet Protocol Version 4, Src: 100.87.227.131, Dst: 100.87.227.140
User Datagram Protocol, Src Port: 47808, Dst Port: 47808
BACnet Virtual Link Control
Building Automation and Control Network NPDU
Building Automation and Control Network APDU
    0101 .... = APDU Type: Error (5)
    Error Class: property (2)
    Error Code: unknown-property (32)
        Application Tag: Enumerated, Length/Value/Type: 1

```

Would it be possible to make BAC0 skip this error and continue discovering the other points and not fail hard?

Commenting out the Raise statement in the `Read.py` file I get this other bacpypes error:

```
  File "/home/deviceenrolment/.local/lib/python3.9/site-packages/bacpypes/primitivedata.py", line 814, in decode
    raise InvalidTag("real application tag required")
bacpypes.errors.InvalidTag: real application tag required

```

Thank you!
Francesco

pisuke, 2025-01-16:
Hello @ChristianTremblay is there any way I can help with to prioritise this fix? Thank you!

ChristianTremblay, 2025-01-17:
I'm always open to PR. All development will be done on the newest version of BAC0, using bacpypes3.

That said, FYI, units is a required properties on Analog Value object

[image]

We can always be more resilient... but that would be nice if things talking BACnet would conform to the requirements...

ChristianTremblay, 2025-01-17:
I'm not sure if I should just return "noUnit" or return `EngineeringUnits("noUnits")` like I did, so if you can test, that would help me.

Thanks

closed 2025-03-26 by github-actions[bot]