ChristianTremblay/BAC0 #354: Failed to get analogValue by BAC0

zzzertion, 2022-09-28:
Hello everyone, I meet some trouble when reading analogValue by BAC0 and I was thinking anyone could help me.

1. We are using a Raspberry Pi to talk with a router, and the BACnet controller is connecting with this router.
I am much familiar with BACnet tool, and I successfully get the data I want by  **_./bin/bacrp 2039307 2 86 85_**
So I think the connection is good, and my Device ID is 2039307, 2 means analogValue,  86  means Object ID and 85 means current value.

2. Then I move to BAC0. 
I am running 

**_import BAC0
myIPAddr = '10.192.62.18/24'
bacnet = BAC0.connect(ip=myIPAddr)
point = bacnet.read('10.192.62.15/24 analogValue 86 presentValue')
print(point)_**

which returns me an error:
**_2022-09-28 16:56:03,582 - WARNING | Unknown object 10.192.62.15/24 analogValue 86 presentValue
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/pi/PycharmProjects/Cornell_Occ/BACread.py", line 6, in <module>
    point = bacnet.read('10.192.62.15/24 analogValue 86 presentValue')
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/BAC0/core/io/Read.py", line 209, in read
    raise UnknownObjectError("Unknown object {}".format(args))
BAC0.core.io.IOExceptions.UnknownObjectError: Unknown object 10.192.62.15/24 analogValue 86 presentValue_**

I also tried 
**_import BAC0
myIPAddr = '10.192.62.18/24'
bacnet = BAC0.connect(ip=myIPAddr)
bacnet.read('10.192.62.15/24 device 20393007 objectList')_**

which returns me an error:
**_confirmed request encoding error: ValueError('instance number out of range')
Traceback (most recent call last):
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/bacpypes/appservice.py", line 1486, in sap_indication
    apdu.encode(xpdu)
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/bacpypes/apdu.py", line 695, in encode
    Sequence.encode(self, self._tag_list)
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/bacpypes/constructeddata.py", line 111, in encode
    helper = element.klass(value)
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 1743, in __init__
    self.set_tuple(*arg)
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/bacpypes/primitivedata.py", line 1767, in set_tuple
    raise ValueError("instance number out of range")
ValueError: instance number out of range
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/pi/PycharmProjects/Cornell_Occ/BACread.py", line 5, in <module>
    bacnet.read('10.192.62.15/24 device 20393007 objectList')
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/BAC0/core/io/Read.py", line 219, in read
    "APDU Abort Reason : {}".format(reason)
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : Timeout_**

I also tried:
mycontroller = BAC0.device('10.192.62.15/24 , 2039307, bacnet)
mycontroller.points
which also failed. 

I also tried:
**_import BAC0
myIPAddr = '192.168.1.10/24'
bacnet = BAC0.connect(ip = myIPAddr)
bacnet.read('2:5 analogValue 86 presentValue')_**

But I don't know how to replace 2:5.
From other posters and Manuel, this should be IP address. But why can we use 2:5

Below is the results for bacnet.discover()

**_2022-09-28 17:01:15,720 - WARNING | 128.253.109.166 Rejected message to network (reason : It is an unknown network layer message)
2022-09-28 17:01:16,006 - INFO    | 128.253.109.243 network number is 5
2022-09-28 17:01:16,015 - INFO    | 128.253.109.61 network number is 5
2022-09-28 17:01:16,025 - INFO    | 128.253.109.111 network number is 5
2022-09-28 17:01:16,031 - INFO    | 128.253.109.88 network number is 5
2022-09-28 17:01:16,038 - INFO    | 128.253.109.194 network number is 5
2022-09-28 17:01:16,045 - INFO    | 128.253.109.77 network number is 5
2022-09-28 17:01:16,047 - INFO    | 128.253.109.64 network number is 5
2022-09-28 17:01:16,053 - INFO    | 128.253.109.82 network number is 5
2022-09-28 17:01:16,059 - INFO    | 128.253.109.62 network number is 5
2022-09-28 17:01:16,067 - INFO    | 128.253.109.73 network number is 5
2022-09-28 17:01:16,071 - INFO    | 128.253.109.191 network number is 5
2022-09-28 17:01:16,075 - INFO    | 128.253.109.84 network number is 5
2022-09-28 17:01:16,080 - INFO    | 128.253.109.93 network number is 5
2022-09-28 17:01:16,082 - INFO    | 128.253.109.56 network number is 5
2022-09-28 17:01:16,084 - INFO    | 128.253.109.68 network number is 1041
2022-09-28 17:01:17,674 - INFO    | 10.192.62.15 router to [20391, 20392, 20394, 20396, 20395, 20393]
2022-09-28 17:01:17,681 - INFO    | 128.253.109.254 router to [10]
2022-09-28 17:01:17,720 - INFO    | 128.253.109.28 router to [90]
2022-09-28 17:01:19,670 - INFO    | Addr : 128.253.109.28
2022-09-28 17:01:21,673 - INFO    | Found those networks : {5, 20391, 20392, 20393, 20394, 20395, 20396, 10, 1041, 90}
2022-09-28 17:01:21,673 - INFO    | Discovering network 5
2022-09-28 17:01:24,678 - INFO    | Discovering network 20391
2022-09-28 17:01:27,684 - INFO    | Discovering network 20392
2022-09-28 17:01:30,689 - INFO    | Discovering network 20393
2022-09-28 17:01:33,695 - INFO    | Discovering network 20394
2022-09-28 17:01:36,698 - INFO    | Discovering network 20395
2022-09-28 17:01:39,702 - INFO    | Discovering network 20396
2022-09-28 17:01:42,707 - INFO    | Discovering network 10
2022-09-28 17:01:45,712 - INFO    | Discovering network 1041
2022-09-28 17:01:48,719 - INFO    | Discovering network 90_**

And I think the Device ID 2039307 should belong to network 20393, and belong to router 10.192.62.15.
I tried all the combination I can think, but all failed. So I was wondering anyone could give some direction.

Thank you very much!

zzzertion, 2022-09-28:
@ChristianTremblay Hi Sir...... I know you are super super excellent in this field...... Could you help me take a look when you have time, I would be very very grateful !!!

zzzertion, 2022-09-29:
> Try removing the subnet in your request. OInly use subnet when you define your bacnet network.
> 
> ```python
> import BAC0
> myIPAddr = '10.192.62.18/24'
> bacnet = BAC0.connect(ip=myIPAddr)
> point = bacnet.read('10.192.62.15 analogValue 86 presentValue')
> print(point)
> ```
> 
> and so on...

Thank you for your kind, prompt reply!!!

but.... still error like this:
2022-09-28 21:36:34,772 - WARNING | Unknown object 10.192.62.15 analogValue 86 presentValue
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/pi/Downloads/pycharm-community-2020.3.3/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/pi/PycharmProjects/Cornell_Occ/BACread.py", line 4, in <module>
    point = bacnet.read('10.192.62.15 analogValue 86 presentValue')
  File "/home/pi/PycharmProjects/Cornell_Occ/venv/lib/python3.7/site-packages/BAC0/core/io/Read.py", line 209, in read
    raise UnknownObjectError("Unknown object {}".format(args))
BAC0.core.io.IOExceptions.UnknownObjectError: Unknown object 10.192.62.15 analogValue 86 presentValue

import BAC0
myIPAddr = '10.192.62.18/24'
bacnet = BAC0.connect(ip=myIPAddr)
point = bacnet.read('10.192.62.15 analogValue 86 presentValue')
print(point)

if I change 86 to something like 2, it can give me a reading.. But what I need is 86.... I guess because there is a router in my case...make it complex??

ChristianTremblay, 2022-09-29:
what is the result of

```python
bacnet.discover()
``

If you are using a BACnet/IP -> MSTP router, you will probably end up knowing on which BACnet network you find the device with device ID 2039307

Then you will be able to point to the address of this device, instead of pointing to the router itself.

example

```python
bacnet.read('2:4 analogValue 86 presentValue')
```

Reading AV 86 on a controller with a MAC adress of 4 in the MSTP network 2

zzzertion, 2022-09-29:
Exactly!! I think this is the key point! This is my first time use BACnet through a router, and I am still confused about the 
**_"MAC adress of 4 in the MSTP network 2"_**
I am nor sure where I can get this information.

Below is the results for discovver:

2022-09-28 21:47:33,031 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2022-09-28 21:47:33,031 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-09-28 21:47:33,032 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-09-28 21:47:33,033 - INFO    | Starting TaskManager
2022-09-28 21:47:33,034 - INFO    | Using ip : 10.192.62.18
2022-09-28 21:47:33,115 - INFO    | Starting app...
2022-09-28 21:47:33,116 - INFO    | BAC0 started
2022-09-28 21:47:33,117 - INFO    | Registered as Simple BACnet/IP App
2022-09-28 21:47:33,124 - INFO    | Update Local COV Task started
2022-09-28 21:47:33,190 - WARNING | 128.253.109.166 Rejected message to network (reason : It is an unknown network layer message)
2022-09-28 21:47:33,459 - INFO    | 128.253.109.243 network number is 5
2022-09-28 21:47:33,463 - INFO    | 128.253.109.84 network number is 5
2022-09-28 21:47:33,466 - INFO    | 128.253.109.73 network number is 5
2022-09-28 21:47:33,469 - INFO    | 128.253.109.77 network number is 5
2022-09-28 21:47:33,472 - INFO    | 128.253.109.64 network number is 5
2022-09-28 21:47:33,476 - INFO    | 128.253.109.68 network number is 1041
2022-09-28 21:47:33,480 - INFO    | 128.253.109.82 network number is 5
2022-09-28 21:47:33,485 - INFO    | 128.253.109.191 network number is 5
2022-09-28 21:47:33,490 - INFO    | 128.253.109.56 network number is 5
2022-09-28 21:47:33,493 - INFO    | 128.253.109.62 network number is 5
2022-09-28 21:47:33,496 - INFO    | 128.253.109.111 network number is 5
2022-09-28 21:47:33,499 - INFO    | 128.253.109.61 network number is 5
2022-09-28 21:47:33,502 - INFO    | 128.253.109.194 network number is 5
2022-09-28 21:47:33,506 - INFO    | 128.253.109.88 network number is 5
2022-09-28 21:47:33,510 - INFO    | 128.253.109.93 network number is 5
2022-09-28 21:47:35,131 - INFO    | 10.192.62.15 router to [20391, 20392, 20394, 20396, 20395, 20393]
2022-09-28 21:47:35,138 - INFO    | 128.253.109.254 router to [10]
2022-09-28 21:47:35,174 - INFO    | 128.253.109.28 router to [90]
2022-09-28 21:47:37,129 - INFO    | Addr : 128.253.109.28
2022-09-28 21:47:39,131 - INFO    | Found those networks : {5, 20391, 20392, 20393, 20394, 20395, 20396, 10, 1041, 90}
2022-09-28 21:47:39,132 - INFO    | Discovering network 5
2022-09-28 21:47:42,138 - INFO    | Discovering network 20391
2022-09-28 21:47:45,143 - INFO    | Discovering network 20392
2022-09-28 21:47:48,149 - INFO    | Discovering network 20393
2022-09-28 21:47:51,156 - INFO    | Discovering network 20394
2022-09-28 21:47:54,163 - INFO    | Discovering network 20395
2022-09-28 21:47:57,169 - INFO    | Discovering network 20396
2022-09-28 21:48:00,176 - INFO    | Discovering network 10
2022-09-28 21:48:03,183 - INFO    | Discovering network 1041
2022-09-28 21:48:06,189 - INFO    | Discovering network 90

Thank you!

zzzertion, 2022-09-29:
Hey!!!!!!!!!!!!!!!!!!!!!!!!!!!
Thanks for the very important hint!!!

import BAC0
myIPAddr = '10.192.62.18/24'
bacnet = BAC0.connect(ip=myIPAddr)
point = bacnet.read('20393:7 analogValue 86 presentValue')
print(point)

I tried this one and worked!!!

zzzertion, 2022-09-29:
I start to understand the meaning of 2:5, COOL! I will try to overwrite next step!
Thank you!!!!!!!

zzzertion, 2022-09-29:
wow, sad.... Trouble again...

I am using 
**_mycontroller['point_name'] = 23_**
and
**_bacnet.write('address object object_instance property value - priority')_**

No error but the value didn't change....
I can change it by bacwp in bacnet tool.... So I am confused again.......

zzzertion, 2022-09-29:
Hi Sir, thanks for your kind help. 
I use same priority as I used in bacnet tool.
but
**mycontroller['point']=15.0** 
can't help me to overwrite;

**prop = ('analogValue',86,'presentValue')
mycontroller.write_property(prop,value=8,priority=8)**
this one can help me to overwrite.

Another interested thing is after I run command above, the value changed if I call bacrp in bacnet tool,
but if I call mycontroller['point'], it won't provide any update.
I have to call mycontroller = BAC0.device('20393:7', 2039307, bacnet) again
and run mycontroller['point'] again to get the updated value.

Is that normal? Does that mean I have to re call the controller again to get the updated value?
Like I want to grab data every 5 minutes, should I do
1) print( mycontroller['point'])

or

2) mycontroller = BAC0.device('20393:7', 2039307, bacnet)
    print( mycontroller['point'])

Thank you for your help!

ChristianTremblay, 2022-09-29:
Controller["point"] = x

Will write at priority 16 by default if the object is a AnalogValue. 

BAC0 will use priority 8 for analog output or binary output as you will typically override outputs of a controller. By opposition to values which can be set. 

Controller["point"].write(value, priority=8) will work

zzzertion, 2022-09-29:
OMG This is working! And I don't need to call controller again! Awesome!
Thank you so much otherwise it will take me 10000000000 years.

closed 2022-12-07 by github-actions[bot]