ChristianTremblay/BAC0 #355: cannot connect to device

Ferkelcode, 2022-10-08:


ChristianTremblay, 2022-11-09:
illegal IP address string passed to inet_aton

this comes from the OS, are you sure you didn't type a wrong IP ?

ChristianTremblay, 2022-11-09:
and yes, if a software already use port 47808 on a network interface, the socker is used and BAC0 won't work.

BAC0 acts as a BACnet device and must be On/alive/using the socket all the time.

closed 2023-01-16 by github-actions[bot]

Ferkelcode, 2023-03-04:
> and yes, if a software already use port 47808 on a network interface, the socker is used and BAC0 won't work.
> 
> BAC0 acts as a BACnet device and must be On/alive/using the socket all the time.

When I first time used the code  `BAC0.connect`, everything worked fine. But now it doesn't work, did I need to disconnect/switch off something that I connected in the first time, to free the prot 47808?

rtarpara, 2024-09-15:
Any help with below error? Not able to connect BAC0. Port 47808 is already free.

runfile('C:/Users/User1/.spyder-py3/temp.py', wdir='C:/Users/User1/.spyder-py3')
[09/14/24 22:41:22] INFO     2024-09-14 22:41:22,066 - INFO    |   notes.py:267
                             Starting Asynchronous BAC0 version                
                             2024.09.10 (Lite)                                 
                    INFO     2024-09-14 22:41:22,069 - INFO    |   notes.py:267
                             Using bacpypes3 version 0.0.98                    
                    INFO     2024-09-14 22:41:22,074 - INFO    |   notes.py:267
                             Use BAC0.log_level to adjust                      
                             verbosity of the app.                             
                    INFO     2024-09-14 22:41:22,078 - INFO    |   notes.py:267
                             Ex. BAC0.log_level('silence') or                  
                             BAC0.log_level('error')                           
                    INFO     2024-09-14 22:41:22,091 - INFO    |    Lite.py:164
                             Using ip : 192.168.201.15/24 on port              
                             47808 | broadcast : 192.168.201.255               
Traceback (most recent call last):

  File C:\ProgramData\anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File c:\users\user1\.spyder-py3\temp.py:9
    bacnet = BAC0.connect(ip='192.168.201.15/24')

  File C:\ProgramData\anaconda3\Lib\site-packages\BAC0\scripts\Lite.py:168 in __init__
    Base.__init__(

  File C:\ProgramData\anaconda3\Lib\site-packages\BAC0\scripts\Base.py:130 in __init__
    raise InitializationError(

InitializationError: IP Address provided (192.168.201.15) invalid. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask')

[09/14/24 22:41:22] INFO     2024-09-14 22:41:22,112 - INFO    |   notes.py:267
                             Installing recurring task Ping Task               
                             (id:2542739881952)