ChristianTremblay/BAC0 #357: Cannot write to a BACnet property with a different priority

chipbutty, 2022-11-04:
Hi, 
I am trying to write to a BACnet property with a priority other than '8'. I tried bacnet.write() by following the example in the documentation e.g. `bacnet.write('2:5 analogValue 1 presentValue 100 - 8')`
e.g. [https://bac0.readthedocs.io/en/latest/BAC0.core.io.html]

I get the following error in parsing the argument: 
args = ['39', 'presentValue', '75.0', '-', '8']

    def _parse_wp_args(self, args):
        """
        Utility to parse the string of the request.
        Supports @obj_ and @prop_ syntax for objest type and property id, useful with proprietary objects and properties.
        """
        if not isinstance(args, list):
            args = args.split()
        obj_type, obj_inst, prop_id = args[:3]
        if obj_type.isdigit():
            obj_type = int(obj_type)
        elif "@obj_" in obj_type:
            obj_type = int(obj_type.split("_")[1])
>       obj_inst = int(obj_inst)
E       ValueError: invalid literal for int() with base 10: 'presentValue'

I also came across this documentation: [device.write_property](https://bac0.readthedocs.io/en/latest/controller.html#bacnet-properties) but have not managed to get it to work. 

Is there a way to use the bac0 device instance e.g. controller object to write to a property with varying priorities?

chipbutty, 2022-11-04:
After much debugging, it seems that the order in which the parameters need to be sent do not match what's in the documentation. The input parameter string is : 'network addr:addr object_type object_instance property_id value - priority'
While it detects everything correct with the above info, including device instance etc. 

`self = Bacnet Network using ip 192.168.1.153 with device id 3056963 | Featuring Bokeh and Pandas
args = ['153:199', 'analogValue', '39', 'presentValue', '75.0', '-', ...]
vendor_id = 0, timeout = 10

    def write(self, args, vendor_id=0, timeout=10):
        """Build a WriteProperty request, wait for an answer, and return status [True if ok, False if not].
    
        :param args: String with <addr> <type> <inst> <prop> <value> [ <indx> ] - [ <priority> ]
        :returns: return status [True if ok, False if not]
    
        *Example*::
    
            import BAC0
            bacnet = BAC0.lite()
            bacnet.write('2:5 analogValue 1 presentValue 100 - 8')
    
        Direct the controller at (Network 2, address 5) to write 100 to the presentValues of
        its analogValue 1 (AV:1) at priority 8
        """
        if not self._started:
            raise ApplicationNotStarted("BACnet stack not running - use startApp()")
        args = args.split()
        self.log_title("Write property", args)
    
        try:
            # build a WriteProperty request
            iocb = IOCB(self.build_wp_request(args, vendor_id=vendor_id))
            iocb.set_timeout(timeout)
            # pass to the BACnet stack
            deferred(self.this_application.request_io, iocb)
            self._log.debug("{:<20} {!r}".format("iocb", iocb))
    
        except WritePropertyException as error:
            # construction error
            self._log.exception("exception: {!r}".format(error))
    
        iocb.wait()  # Wait for BACnet response
    
        if iocb.ioResponse:  # successful response
            apdu = iocb.ioResponse
    
            if not isinstance(apdu, SimpleAckPDU):  # expect an ACK
                self._log.warning("Not an ack, see debug for more infos.")
                self._log.debug(
                    "Not an ack. | APDU : {} / {}".format((apdu, type(apdu)))
                )
                return
    
        if iocb.ioError:  # unsuccessful: error/reject/abort
            apdu = iocb.ioError
            reason = find_reason(apdu)
>           raise NoResponseFromController("APDU Abort Reason : {}".format(reason))
E           BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : Timeout

..\venv\lib\site-packages\BAC0\core\io\Write.py:113: NoResponseFromController`

chipbutty, 2022-11-04:
I am still puzzled why we cannot use the defined bacnet device instance e.g. controller object instead of re-searching for the device by giving the network Ip and device ip.

ChristianTremblay, 2022-11-04:
Have you tried

dev["pointname"].write(value, priority=10)

?

chipbutty, 2022-11-07:
I tried, but there is no option 'write' for the bacnet device instance.

ChristianTremblay, 2022-11-08:
Device instance is not something I usually change using BACnet. In fact, a lot of device don't support writing to device instance property.

chipbutty, 2022-11-08:
In that case, how should I define 'dev'  in the following statement - dev["pointname"].write(value, priority=10)

chipbutty, 2022-11-08:
ok, I think I got it to work. Now, how do release the priority and get it back to the original value?

ChristianTremblay, 2022-11-09:
write 'null' at the same priority

```
dev['point'].write("null", priority=8)
```

chipbutty, 2022-11-09:
Not sure if I am doing it correctly. 

`val01 = controller['Standby Time']`
`print(val01)`
`controller['Standby Time'].write(value=1.5, priority=10)`
`val02 = controller['Standby Time']`
`print(val02)`
`controller['Standby Time'].write(value='null', priority=10)`
`val03 = controller['Standby Time']`
`print(val03)`

the above code returns: 
`SXWSE9650Bxx-86255/Standby Time : 0.50 hours`
`SXWSE9650Bxx-86255/Standby Time : 1.50 hours`
`SXWSE9650Bxx-86255/Standby Time : **1.50 hours**`

The value does not get reset to val01 (0.50)

Perhaps I am not doing it right.

ChristianTremblay, 2022-11-09:
I think everything is ok. It is probably because the variable is set internally by your controller, and as soon as you wrote @10, you changed the value.

```
controller['Standby Time'].priority
```
will help you understand what is going on with this object before writing to it.

chipbutty, 2022-11-09:
Is there a way to read the relinquished value after the priority is released?

ChristianTremblay, 2022-11-28:
```python
# read it
d['TUNING-RESET'].read_property(prop='relinquishDefault')

# write to it
d['TUNING-RESET'].default('inactive')
```

closed 2023-01-30 by chipbutty