ChristianTremblay/BAC0 #358: Cannot create two instances at the same time

dqcontrols, 2022-11-08:
Hi, 

I am looking to create multiple RTUs (Rooftop Units) on a single Raspberry Pi as BACnet devices. I am running into the following error:

BAC0.core.io.IOExceptions.InitializationError: IP Address (10.0.0.10) already used by BAC0. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask').

The raspberry Pi is on a Wi-Fi network that has the following ip address (10.0.0.10). 

The code I am running is as follows:

import BAC0
bacnet1 = BAC0.lite()
bacnet2 = BAC0.lite(ip = '10.0.0.10/24')

The error message is as follows:
Python 3.9.2 (/usr/bin/python3)
>>> %Run multiple_devices_test.py
2022-11-08 13:42:29,989 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2022-11-08 13:42:29,990 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-08 13:42:29,991 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-08 13:42:29,991 - INFO    | Starting TaskManager
2022-11-08 13:42:30,091 - INFO    | Using ip : 10.0.0.10
2022-11-08 13:42:30,202 - INFO    | Starting app...
2022-11-08 13:42:30,204 - INFO    | BAC0 started
2022-11-08 13:42:30,205 - INFO    | Registered as Simple BACnet/IP App
2022-11-08 13:42:30,213 - INFO    | Update Local COV Task started
2022-11-08 13:42:30,214 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2022-11-08 13:42:30,214 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-08 13:42:30,215 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-08 13:42:30,216 - INFO    | Using ip : 10.0.0.10
Traceback (most recent call last):
  File "/home/demandq/Desktop/multiple_devices_test.py", line 4, in <module>
    bacnet2 = BAC0.lite(ip = '10.0.0.10/24')
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/scripts/Lite.py", line 152, in __init__
    Base.__init__(
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/scripts/Base.py", line 138, in __init__
    raise InitializationError(
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (10.0.0.10) already used by BAC0. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask')
>>>

Any help would be appreciated. Thanks in advance!

ChristianTremblay, 2022-11-09:
You must use different IP addresses. Your using the same. 
It is possible to give multiple IPs to the same NIC.

dqcontrols, 2022-11-09:
> 

Hi Christian,

Thanks for the response. I did some research into how to set up multiple IPs to the Raspberry Pi wlan0. Here is the current ip address settings in my Pi, with the ip addresses in use in **bold**

demandq@raspberrypi:~ $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:dc:87:37 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:89:d2:62 brd ff:ff:ff:ff:ff:ff
    **inet 10.0.0.10/24 brd 10.0.0.255 scope global dynamic noprefixroute wlan0**
       valid_lft 86337sec preferred_lft 75537sec
    **inet 11.0.0.100/24 scope global wlan0**
       valid_lft forever preferred_lft forever
    inet6 fe80::27c2:c87b:fcea:dc5f/64 scope link
       valid_lft forever preferred_lft forever

My current code is:
bacnet = BAC0.connect(ip='10.0.0.10/24')
bacnet1 = BAC0.conenct(ip='11.0.0.100/24')

This should connect each instance shown above to a different IP address. However, this still runs into issues. Any insights? I do believe that the ip addresses in the Pi are being set correctly as far as I am aware.

The new error I encounter is as follows:
Python 3.9.2 (/usr/bin/python3)
>>> %Run RTU_sim_N4_BAC0.py
2022-11-09 14:10:01,422 - INFO    | Starting BAC0 version 22.9.21 (Complete)
2022-11-09 14:10:01,423 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-09 14:10:01,425 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-09 14:10:01,427 - INFO    | Starting TaskManager
2022-11-09 14:10:01,432 - INFO    | Using ip : 10.0.0.10
2022-11-09 14:10:01,596 - INFO    | Starting app...
2022-11-09 14:10:01,598 - INFO    | BAC0 started
2022-11-09 14:10:01,599 - INFO    | Registered as Simple BACnet/IP App
2022-11-09 14:10:01,607 - INFO    | Update Local COV Task started
2022-11-09 14:10:04,132 - INFO    | Server started : http://10.0.0.10:8111
 * Serving Flask app "BAC0.web.FlaskServer" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2022-11-09 14:10:04,251 - INFO    | 10.0.0.101 network number is 1000
2022-11-09 14:10:08,144 - INFO    | Found those networks : {1000}
2022-11-09 14:10:08,146 - INFO    | Discovering network 1000
2022-11-09 14:10:11,155 - INFO    | Starting BAC0 version 22.9.21 (Complete)
2022-11-09 14:10:11,157 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-09 14:10:11,158 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-09 14:10:11,159 - INFO    | Using ip : 11.0.0.100
2022-11-09 14:10:11,174 - INFO    | Starting app...
2022-11-09 14:10:11,178 - INFO    | BAC0 started
2022-11-09 14:10:11,179 - INFO    | Registered as Simple BACnet/IP App
2022-11-09 14:10:11,190 - INFO    | Update Local COV Task started
an error has occurred: no adapters
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/core.py", line 165, in run
    fn(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/service/device.py", line 32, in startup
    self.i_am()
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/service/device.py", line 133, in i_am
    self.request(iAm)
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/core/app/ScriptApplication.py", line 246, in request
    super(BAC0Application, self).request(apdu)
[... 116 lines cut ...]
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.9/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.9/http/server.py", line 138, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.9/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/web/FlaskServer.py", line 42, in run
    self.process()
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/web/FlaskServer.py", line 46, in process
    self.task()
  File "/home/demandq/.local/lib/python3.9/site-packages/BAC0/web/FlaskServer.py", line 165, in task
    self._log.warning("Flask server already running", err)
AttributeError: 'FlaskServer' object has no attribute '_log'
2022-11-09 14:10:17,690 - INFO    | Found those networks : set()
2022-11-09 14:10:17,692 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
>>>

ChristianTremblay, 2022-11-12:
Use BAC0.lite instead of BAC0.connect

closed 2023-01-19 by github-actions[bot]