ChristianTremblay/BAC0 #359: Is there any way we can change the device instance number?

zzzertion, 2022-11-09:
Hi, I am using Raspberry PI to read/write data, everything goes on well.
But the EMS guys ask me to change the device instance number of my Raspberry PI.
I never did this before... I know how to read/write data for like an AHU, but I have no idea how to change the name for Pi.....
I am not sure can we do that by BAC0 package?

Thanks!!

bbartling, 2022-11-09:
Can you show your code how you define your BAC0 device there is a way to do this.

zzzertion, 2022-11-09:
Hi! Thanks for your kind reply.
I am using:

_**import BAC0
myIPAddr = '10.192.62.17/24'
bacnet = BAC0.connect(ip=myIPAddr)
Mycontroller = *****
Mycontroller .read()**_

Sorry I just lose connection with my Pi, but I remember I am using this type of framework.
Now the Pi has instance number of 3056631 they want me to switch to 2039002.

bbartling, 2022-11-09:
something like

`bacnet = BAC0.connect(ip=myIPAddr,instance=2039002)`

but I could be way off its been a while

zzzertion, 2022-11-09:
Sad.. 

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-060e2892eeaa> in <module>()
      1 myIPAddr = '10.192.62.17/24'
----> 2 bacnet = BAC0.connect(ip=myIPAddr, instance=2039002)

~/berryconda3/envs/jl/lib/python3.6/site-packages/BAC0/scripts/Lite.py in __init__(self, ip, port, mask, bbmdAddress, bbmdTTL, bdtable, ping, ping_delay, db_params, **params)
    156             bbmdTTL=bbmdTTL,
    157             bdtable=bdtable,
--> 158             **params
    159         )
    160 

TypeError: __init__() got an unexpected keyword argument 'instance'

bbartling, 2022-11-09:
May have to wait until @Christian Tremblay
***@***.***> reply
s I can't remember

ChristianTremblay, 2022-11-09:
you were close
```
bacnet = BAC0.connect(ip=myIPAddr,deviceID=2039002)
```

you can also change the name

```
bacnet = BAC0.connect(ip=myIPAddr,deviceID=2039002, localObjName="Whatever")
```

zzzertion, 2022-11-09:
Thanks for your help.. But it seems something wrong...

ChristianTremblay, 2022-11-09:
typo.... `deviceId`

zzzertion, 2022-11-09:
It works.................................... Thank you so much for helping me again!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

closed 2022-11-28 by ChristianTremblay