ChristianTremblay/BAC0 #360: Find present objects in connected BacNet Device

dheerajdevalekar, 2022-11-17:
Hello,
I used YABE(Yet Another Bacnet Explorer) as a BacNet Simulator. I used devices = bacnet.discover(global_broadcast=True) this command to find a connected Devices. I found the Bacnet Devices also. But I need to connect one of them, so how can I do?
Also I need to fetch the present objects in that device...
Can you help in this???

ChristianTremblay, 2022-11-28:
It depends on what you want to do....

The object list of a device can be found by reading it

```python
bacnet.read('address device deviceId objectList')
```

you could also create a `BAC0.device` and get all points...

You should start here https://bac0.readthedocs.io/en/latest/

bbartling, 2022-11-28:
This may get you going if you know your devices and objects thru Yabe...also checkout the `BAC0.connect()` which works a little different I think it uses polling and you can define a database Vs a `BAC0.lite()` app:

```
import BAC0, time, random

bacnet = BAC0.lite()

address = '12345:2'
object_type = 'analogValue'
object_instance = '302'
priority = 10 #BACnet priority

while True:

    # create write statement and run
    write_vals = f'{address} {object_type} {object_instance} presentValue {str(random.randint(66, 122))} - {priority}'
    print("Excecuting write_vals statement:", write_vals)

    write_result = bacnet.write(write_vals)
    print('BACnet written to MSTP device')
    print('Wait a few seconds....')
    time.sleep(5)

    read_vals = f'{address} {object_type} {object_instance} presentValue'
    check = bacnet.read(read_vals)
    print(check)

    # create write release statement and run
    write_rel = f'{address} {object_type} {object_instance} presentValue null - {priority}' 
    print("Excecuting write_vals statement:", write_rel)

    write_result = bacnet.write(write_rel)
    print('BACnet written to MSTP device')
    print('Wait a few seconds....')
    time.sleep(5)

    read_vals2 = f'{address} {object_type} {object_instance} presentValue'
    check2 = bacnet.read(read_vals2)
    print(check2)

bacnet.disconnect()
```

ChristianTremblay, 2022-11-28:
> BAC0.connect() which works a little different I think it uses polling and you can define a database Vs a BAC0.lite()

Connect adds a Bokeh Server + web app to `BAC0.lite`.
Anything else works the same. To be honest, I use lite 99.9% of the time.

closed 2023-02-05 by github-actions[bot]

dheerajdevalekar, 2023-03-02:
Done with this. I did following code:

import BAC0
from time import sleep
from loguru import logger
import json
BAC0.log_level('silence')  # This is to silence the default BacNet Logs

my_ip = "192.168.9.13/24"
bacnet_device_ip_with_port = "192.168.9.150:47808"

bacnet = BAC0.lite(ip=my_ip)

whois = bacnet.whois()
print(f"Whois: {whois}")
print(f"----------------------------------------------------------------")
a = bacnet.discover()
print(f"Discovered Devices: {a} with Length: {len(a)}")
print(f"----------------------------------------------------------------")
b = bacnet.whois(global_broadcast=True)
print(f"Global Broadcast: {b} with Length: {len(b)}")
print(f"----------------------------------------------------------------")

def get_existing_objectlist(net_id, device_id):
    objects = []
    # TODO: To get the Existing ObjectList Present on deviceId
    """
    bacnet.read(f'{loop[0]} device {loop[1]} objectList')
    """
    try:
        objects = bacnet.read(f'{net_id} device {device_id} objectList')
    except Exception as err:
        logger.error(f"{err}")
    finally:
        return objects

def get_bacnet_properties_value():
    try:
        pass
    except Exception as er:
        logger.error(f"{er}")

def get_data():
    length_of_objects = {}
    final_values = {}
    try:
        for loop in a:
            if loop[0] == '192.168.9.150':
                continue
            try:
                net_addr, dev_id = loop[0].split(":")
            except:
                print(f"Error to Get NET ADDRESS: {loop}")
                continue
            # dev_id = loop[1]
            found_objects = get_existing_objectlist(net_id=loop[0], device_id=dev_id)
            all_objects_value = {}
            length_of_objects.update({loop: len(found_objects)})
            for obj in found_objects:
                if obj[0] != "device":
                    try:
                        value = bacnet.read(f'{net_addr}:{dev_id} {obj[0]} {obj[1]} presentValue')
                    except:
                        value = None
                    try:
                        description = bacnet.read(f'{net_addr}:{dev_id} {obj[0]} {obj[1]} description')
                    except:
                        description = None
                    try:
                        units = bacnet.read(f'{net_addr}:{dev_id} {obj[0]} {obj[1]} units')
                    except:
                        units = None
                    all_objects_value.update({f'{obj[0]}:{obj[1]}-{description}': f'{value} {units}'})
            final_values.update({loop: all_objects_value})
    except KeyboardInterrupt:
        if bacnet is not None:
            bacnet.disconnect()
        print(f"Main STOP & BacNet get Disconnected....!!!")
        # break
    except Exception as e:
        print(f"Main Error: {e}")
    finally:
        print(f'LENGTH::::::: {length_of_objects} with dict_len: {len(length_of_objects)}')
        # print(f'Final:: {final_values} with length: {len(final_values)}')
        sleep(2)
        file_obj = open(f'bacnet_obj_numbers.json', 'w')
        file_obj.write(json.dumps(f'{final_values}', indent=4))
        file_obj.close()
        # print(f'Final:: {final_values} with length: {len(final_values)}')

if __name__ == '__main__':
    logger.debug(f'Start to get Object and Properties Values')
    get_data()