ChristianTremblay/BAC0 #361: Disable saving to DB

clavay, 2022-11-17:
Hello,

I try to connect to BACnet devices to read and write values.
I develop a SCADA system ([PyScada](https://github.com/pyscada/PyScada)) that have his own data management (using a DB).
I don't want to BAC0 to store the data.

Here is my code : 
```
import BAC0
server = BAC0.lite(ip="10.154.254.110/24",port=47808)
2022-11-17 09:55:54,291 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2022-11-17 09:55:54,291 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-17 09:55:54,292 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-17 09:55:54,293 - INFO    | Starting TaskManager
2022-11-17 09:55:54,294 - INFO    | Using ip : 10.154.254.110
2022-11-17 09:55:54,369 - INFO    | Starting app...
2022-11-17 09:55:54,370 - INFO    | BAC0 started
2022-11-17 09:55:54,371 - INFO    | Registered as Simple BACnet/IP App
2022-11-17 09:55:54,376 - INFO    | Update Local COV Task started

dev = BAC0.device('10.154.254.11',11, server, history_size=None, poll=0, auto_save=False,)
2022-11-17 10:02:10,299 - INFO    | Changing device state to DeviceDisconnected'>
2022-11-17 10:02:10,318 - INFO    | Changing device state to RPMDeviceConnected'>
2022-11-17 10:02:10,342 - INFO    | Device 11:[504 ModBus] found... building points list
2022-11-17 10:02:10,731 - INFO    | Ready!

dev.disconnect()
2022-11-17 10:07:03,968 - INFO    | Wait while stopping polling
2022-11-17 10:07:03,968 - ERROR   | Pandas is required to save to SQLite.
2022-11-17 10:07:03,969 - INFO    | Changing device state to DeviceFromDB'>
2022-11-17 10:07:03,970 - INFO    | Initializing DB
2022-11-17 10:07:03,970 - INFO    | Missing argument DB
2022-11-17 10:07:03,970 - ERROR   | Problem with DB initialization : Please provide db name using device.load_db('name')
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-9-2ea8f68904a0> in <module>
----> 1 dev.disconnect()

/usr/local/lib/python3.9/dist-packages/BAC0/core/devices/Device.py in disconnect(self, save_on_disconnect, unregister)
    466         if unregister:
    467             self.properties.network.unregister_device(self)
--> 468         self.new_state(DeviceFromDB)
    469 
    470     def connect(self, *, db=None):

/usr/local/lib/python3.9/dist-packages/BAC0/core/devices/Device.py in new_state(self, newstate)
    213         )
    214         self.__class__ = newstate
--> 215         self._init_state()
    216 
    217     def _init_state(self):

/usr/local/lib/python3.9/dist-packages/BAC0/core/devices/Device.py in _init_state(self)
    996     def _init_state(self):
    997         try:
--> 998             self.initialize_device_from_db()
    999         except ValueError as e:
   1000             self._log.error("Problem with DB initialization : {}".format(e))

/usr/local/lib/python3.9/dist-packages/BAC0/core/devices/Device.py in initialize_device_from_db(self)
   1060         else:
   1061             self._log.info("Missing argument DB")
-> 1062             raise ValueError("Please provide db name using device.load_db('name')")
   1063 
   1064         # network = self.properties.network

ValueError: Please provide db name using device.load_db('name')

```
Uncommenting line https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Device.py#L1001
and commenting line https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Device.py#L1002
solve the problem as the device change state to disconnected if init device from db failed.

Am I doing something wrong ?

Thanks

ChristianTremblay, 2022-11-26:
Nothing wrong. Looks and smells like a bug

ChristianTremblay, 2022-11-26:
I think it comes from [here](https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Device.py#L170) where I give an empty string value (`""`) to `db_name` instead of `None`... I'll push an update to the develop branch, if you don't mind to test it when you have time.

ChristianTremblay, 2022-11-26:
It is more than what I first thought... did some experimentation... feedback appreciated

ChristianTremblay, 2022-11-26:
Forgot also...
```
dev1.disconnect(save_on_disconnect=False)
```

clavay, 2022-12-12:
It is ok for me.
Thank you.

closed 2022-12-12 by clavay