ChristianTremblay/BAC0 #362: Can I scan more than 1 BAC0 virtual devices by VOLTTRON

xysedhasan, 2022-11-18:
I have created two virtual devices by(tests/manual_test_create_device.py ) this code on different ports (47808 & 47809) I tried too much to scan devices with VOLTTRON with different scenarios but I can scan only 1 device that is on 47808. If the device is on another port rather than 47808 devices can not be scanned.

bbartling, 2022-11-18:
I use BAC0 with VOLTTRON all the time but always deploy BAC0 on a separate devices like a rasp pi or something. What is the use case for deploying virtual BACnet devices on separate ports on the same device?

ChristianTremblay, 2022-11-26:
By using a different port, you are using a different BACnet network. Discovery will not work over multiple ports. 

You can direct unicast to devices, but anything relying on broadcast won't work

xysedhasan, 2023-01-11:
i got my answer by creating dual ip I am able to do this

closed 2023-01-11 by xysedhasan