ChristianTremblay/BAC0 #363: location property & inactive/active text for binary

kheldaroz402, 2022-11-22:
HI Christian,

how do I set the location property on the device
eg.
`bacnet = BAC0.lite(ip=WebAIConfig['IP'], deviceId=WebAIConfig['BACnetID'], description=WebAIConfig['SystemName'], modelName=WebAIConfig['ControllerType'], firmwareRevision=WebAIConfig['version'],localObjName=WebAIConfig['SystemName'],vendorName='Sphere Systems', vendorId='47')
`
also how do I set the inactive/active text states for a binary object (Note: i am grabbing data from a MySQL DB to create the points)

eg.

```
sql = 'SELECT * FROM WebAI_Program.BACnet_Objects_BinaryValue'
    mycursorProgram.execute(sql)
    mydbProgram.commit()
    myresult = mycursorProgram.fetchall()
    for row in myresult:
        # Supplemental with more details, for demonstration
        states = make_state_text([row['On_lable'], row['Off_lable']])
        if (row['Present_Value'] == 1):
            pv = 'active'
        else:
            pv = 'inactive'
        if (row['Relinquish_Default'] == 1):
            rd = 'active'
        else:
            rd = 'inactive'
        _new_objects = binary_output(
            name=row['Object_Name'],
            instance=row['Object_Identifier'],
            # properties={"stateText": states},
            description=row['Description'],
            presentValue=pv,
            is_commandable=True,
            relinquish_default=rd,
            polarity=row['Polarity'],
            inactive_text=row['Off_lable'],
            active_text=row['On_lable'],

        )

```
I've tried a couple of ways but don't seem to be able to do this in BAC0, I was able to do it BACpypes

ChristianTremblay, 2022-11-26:
Try this

```python
_new_objects = binary_output(
    name=row['Object_Name'],
    instance=row['Object_Identifier'],
    # properties={"stateText": states},
    description=row['Description'],
    presentValue=pv,
    is_commandable=True,
    relinquish_default=rd,
    polarity=row['Polarity'],
    properties={"inactiveText": row['Off_lable'], "activeText": row['On_lable']},
    )
```

kheldaroz402, 2022-11-27:
Thanks Christian,
                that solved the inactive/active text issue

closed 2022-11-27 by kheldaroz402