ChristianTremblay/BAC0 #364: location property

kheldaroz402, 2022-11-27:
HI Christian,

when I try and use the location property it throws the following error

`2022-11-28 08:30:04,581 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-11-28 08:30:04,583 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-11-28 08:30:04,584 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-11-28 08:30:04,584 - INFO    | Starting TaskManager
2022-11-28 08:30:04,587 - INFO    | Using ip : 192.168.2.240
Traceback (most recent call last):
  File "/home/ubuntu/WebAIPy/./Bac0WebAI.py", line 992, in <module>
    main()
  File "/home/ubuntu/WebAIPy/./Bac0WebAI.py", line 960, in main
    WebAI_BACnet_Initialise()
  File "/home/ubuntu/WebAIPy/./Bac0WebAI.py", line 140, in WebAI_BACnet_Initialise
    bacnet = BAC0.lite(ip=WebAIConfig['IP'], deviceId=WebAIConfig['BACnetID'], description=WebAIConfig['SystemName'], location=WebAIConfig['Location'], modelName=WebAIConfig['ControllerType'], firmwareRevision=WebAIConfig['version'],localObjName=WebAIConfig['SystemName'],vendorName='Sphere Systems', vendorId='47')
  File "/home/ubuntu/.local/lib/python3.10/site-packages/BAC0/scripts/Lite.py", line 152, in __init__
    Base.__init__(
TypeError: Base.__init__() got an unexpected keyword argument 'location'
`

code

` bacnet = BAC0.lite(ip=WebAIConfig['IP'], deviceId=WebAIConfig['BACnetID'], description=WebAIConfig['SystemName'], location=WebAIConfig['Location'], modelName=WebAIConfig['ControllerType'], firmwareRevision=WebAIConfig['version'],localObjName=WebAIConfig['SystemName'],vendorName='Sphere Systems', vendorId='47')`

Is location implemented in BAC0? I know it works in BACpypes

ChristianTremblay, 2022-11-28:
Not until tonight...

```python
import BAC0
bacnet = BAC0.lite(location='here')
```
should work

...actually on the develop branch

bbartling, 2022-11-28:
What is the use case of defining "location" in BACnet? I have never heard of that...is it supposed to be like a description type string to show like the street address building number and room of the device?

ChristianTremblay, 2022-11-28:
Same kind of info than description... could be a panel, a room, anything you want

bbartling, 2022-11-28:
From a cybersecurity standpoint does ever seem like risky business? For example if some criminal hacked into some random IP address on the net, run some descriptive BACnet commands....couldnt they then see all this descriptive info and put it together where the building is located or maybe even what the company is....what floor of the building I kind find the important cooling plant to shut down..

ChristianTremblay, 2022-11-28:
Maybe... but if you kept your front door unlock in the first place... you probably didn't hide your IP behind something else anyway... and just by the name of the controller, we'll probably know it's controlling AHU_XYZ or Boiler plant.... do you really care if the location of the controller is in Panel A-32i... or "in the basement" ?

kheldaroz402, 2022-11-28:
After years of working in the HVAC industry (BMS) - the location is extremely important, over time the people who installed the controllers have retired, moved to different companies or the support contract has changed, so a different company is now supporting those controllers. Trying to find controllers or sensors can take hours or even days. Often it can be cheaper to install a new sensor. Ideally the controller should be in the mechanical board, but I've seen them installed in roof cavities, cupboards in carparks, inside the package unit, so just the name of the controller is not always going to tell you where it is.

kheldaroz402, 2022-11-30:
thanks Christian
now that I'm using 22.09.21 location is now working :)

closed 2022-11-30 by kheldaroz402