ChristianTremblay/BAC0 #367: callback function

kheldaroz402, 2022-11-30:
HI Christian,

The next issue that I'm having is with the callback function.

it doesn't appear to be working - I've probably done something wrong

`    for row in myresult:
        bacnet.cov(row['Address'], row['Object_Identifer'], row['BACnet_Object_ID'],  lifetime=row['COV_lifetime'], callback=my_callback)`

[image]

Callback function
```
# The Notification will pass a variable named "elements" to the callback
# your function must include this argument

# elements is a dict containing all the information of the COV

def my_callback(elements):
    print(color.YELLOW + "Present value is : {}".format(elements['properties']['presentValue']) + color.END)
    print(color.PURPLE + "Address is : {}".format(elements['address']) + color.END)
    print(color.GREEN + "ObjectID value is : {}".format(elements['objectID']) + color.END)
    print(color.CYAN + "source value is : {}".format(elements['source']) + color.END)
```

it looks like the COV's are working just not the callback

ChristianTremblay, 2023-01-02:
Hi ! Back from a little screen break, I'll have a look to your case ASAP

kheldaroz402, 2023-02-01:
Hi Christian,
                  Any progress ?

ChristianTremblay, 2023-02-04:
Changing the value, I get the COV back

[image]

closed 2023-04-13 by github-actions[bot]