ChristianTremblay/BAC0 #374: The read and write program works with BAC0, but there is a problem with the YABE connection

KamyabHusain, 2023-01-13:
Whenever I start simulator it starts writing data, and when I run the read program in another window it reads successfully. Now when I try to connect with YABE it shows the below error. Could you please help me with that.

[image]

bbartling, 2023-01-13:
Make sure your BAC0 device is running on a different computer...like I don't think you want to run your bacnet scan tool like Yabe and BAC0 on same device as BACnet UDP port 47808 can only be consumed by one process and they don't take turns using it. I do all my test bench stuff with 2 separate devices

KamyabHusain, 2023-01-16:
Thanks for the help.
it is working with different computer

closed 2023-01-16 by KamyabHusain