ChristianTremblay/BAC0 #379: What is the Network and Address in BacNet over Ethernet

dheerajdevalekar, 2023-02-24:
- What is the Network and Address in BacNet over Ethernet.

- If a Python program is connected to BacNet Router, and Router has many connected Devices. Then how we can accress that device's object along with its Properties in Python....

bbartling, 2023-02-24:
I think you want to ask in your first bullet point BACnet over IP, or BACnet/IP. BACnet over ethernet I don't think industry uses in present times but only BACnet/IP is very common or BACnet or MSTP which is a 2 wire twisted pair network.

I have a BACnet router and BACnet MSTP device on a network on a test bench at my home. I can run this program which is attempting to read hardware address 2 on MSTP network 12345 present value of the temp sensor analogInput 2.

```
import BAC0

bacnet = BAC0.lite()
sensor = bacnet.read('12345:2 analogInput 2 presentValue')
print(sensor)
```
At least if you are on a LAN and everything on the same LAN you never have to worry about anything about a BACnet router all that is handled under the hood.

This program will print:
```
2023-02-24 08:21:42,299 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2023-02-24 08:21:42,303 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-02-24 08:21:42,305 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-02-24 08:21:42,308 - INFO    | Starting TaskManager
2023-02-24 08:21:42,335 - WARNING | Netifaces not installed on your system. BAC0 can't detect the subnet.
Please provide subnet for now, we'll consider 255.255.255.0 (/24).
You can install netifaces using 'pip install netifaces'.
2023-02-24 08:21:42,340 - INFO    | Using ip : 192.168.0.104
2023-02-24 08:21:42,822 - INFO    | Starting app...
2023-02-24 08:21:42,827 - INFO    | BAC0 started
2023-02-24 08:21:42,829 - INFO    | Registered as Simple BACnet/IP App
2023-02-24 08:21:42,852 - INFO    | Update Local COV Task started
2023-02-24 08:21:42,927 - INFO    | 192.168.0.190 router to [12345]
95.72999572753906
```

If you want to expand on reading or writing to lots of devices just expand your program in a synchronous way just make the program longer and longer with more devices/sensors.

I would highly suggest using some [BACnet scan tool like this](https://www.ccontrols.com/sd/bdt.htm) which is free before getting stuff to work in Python.

bbartling, 2023-02-24:
If was device was an IP device just plop in the IP address as a string to the program for example:

import BAC0

bacnet = BAC0.lite()
sensor = bacnet.read('192.168.0.100 analogInput 2 presentValue')
print(sensor)

ChristianTremblay, 2023-03-05:
For BACnet Ethernet devices, you will see something like Network:MAC (ex. `100:00e345ff00cc`)
You then know that this network number is 100...

Delta controllers here are regularly using ethernet as it allows using ethernet network, without the obligation to setup IP addresses.

closed 2023-05-12 by github-actions[bot]