ChristianTremblay/BAC0 #380: writeMultiple fails with "bacpypes.errors.InvalidTag: errorType expected opening tag 0"

CrisCodesCrap, 2023-02-26:
When I try executing the ```writeMultiple``` method the process throws the following error:
[image]
I even tried a minimal reproduction with the following setup:
```
*requirements.txt*
BAC0==22.9.21
bacpypes==0.18.6
colorama==0.4.6
netifaces==0.11.0
pytz==2022.7.1
```

```
*main.py*
from BAC0 import lite

bac = lite()

bac.writeMultiple("192.168.0.128:50286", ['analogValue 1 presentValue 100'])
bac.readMultiple()
```
My python version is 3.11
[image]

Any help would be greatly appreciated!

CrisCodesCrap, 2023-02-26:
Tried it with 
[image]
 and the same setup as before.
Does not work as well.

ChristianTremblay, 2023-03-05:
I'm concerned about the `bac.readMultiple()` part

CrisCodesCrap, 2023-03-05:
This is a mistake in the example.
It does not work either way, any ideas on how we can tackle it?

ChristianTremblay, 2023-03-05:
Ok. 
And this BACnet device is running on port 50286 ? Unusual 

Try keeping only the IP.

CrisCodesCrap, 2023-03-05:
The device running on port 50286 is a virtual device from Yabe's add-ons.
From what I think the issue may be is regarding the parser of bacpypes and how it handles arrays of strings for this specific method.

ChristianTremblay, 2023-03-06:
Array of strings ?

Strings are weird in BACnet,.... I haven't work on writeMultiple for string values but if you look [here](https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/functions/Text.py) you will see tools to write text value in BACnet.

This is used by the `update_description` method of devices....[here](https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Device.py#L795)

CrisCodesCrap, 2023-03-06:
I am talking about the writeMultiple method on this page https://bac0.readthedocs.io/en/latest/write.html

CrisCodesCrap, 2023-03-08:
I'll test the dev branch as soon as I get access to my machine

closed 2023-05-15 by github-actions[bot]