ChristianTremblay/BAC0 #381: Fake air handling unit

bbartling, 2023-03-02:
Would anyone know how to make a more life like fake air handling unit where the point data could represent something close to real & actual???

This script below whips up the points and the discoverable BACnet BAC0 device but its just random data....it would be real cool to make something more life like and I am sure loads of engineering tools exist in Python somewhere I am just not a real expert in Newtons cooling laws but can probably get pretty close. For example just randomly coming up with an outside temperature one can assume what happens to the AHU return, mix, and supply air temperatures....valves and not to mention economizer for typical BAS setpoints...if there are any ideas out there feel free to suggest!

```
import BAC0
import time
import random

from BAC0.core.devices.local.models import (
    analog_output,
    analog_value,
    binary_input,
    analog_input,
    binary_output
)

from BAC0.tasks.RecurringTask import RecurringTask
from bacpypes.primitivedata import Real

# should work on Linux or Windows run with
# $ python g36_fake_ahu.py

# G36 AHU required input points
points = [
    "DAP-SP",
    "DAP-P",
    "SA-T",
    "SAT-SP",
    "MA-T",
    "RA-T",
    "SF-O",
    "HTG-O",
    "CLG-O",
    "SF-C",
    "SF-S",
    "SA-FLOW"]

def changeValues():
    for i in range(len(points)):
        _object = points[i]
        point = bacnet.this_application.get_object_name(_object)
        value = random.randrange(5, 95, 1)
        point.presentValue = Real(value)
        
        # perform a check that it took a new pv value
        bacnet._log.info(f'{_object} is {point.presentValue}')

# create AHU duct pressure setpoint point
_new_objects = analog_value(
    name="DAP-SP",
    properties={"units": "inchesOfWater"},
    description="AHU Duct Pressure Setpoint",
    presentValue=1, is_commandable=True
)

# create AHU duct pressure
_new_objects = analog_input(
    name="DAP-P",
    properties={"units": "inchesOfWater"},
    description="AHU Duct Pressure",
    presentValue=1, is_commandable=False
)

# create AHU supply temp
_new_objects = analog_input(
    name="SA-T",
    properties={"units": "degreesFahrenheit"},
    description="AHU Duct Supply Temp",
    presentValue=75, is_commandable=False
)

# create AHU supply temp setpoint
_new_objects = analog_value(
    name="SAT-SP",
    properties={"units": "degreesFahrenheit"},
    description="AHU Duct Supply Temp",
    presentValue=75, is_commandable=True
)

# create AHU mix temp
_new_objects = analog_input(
    name="MA-T",
    properties={"units": "degreesFahrenheit"},
    description="AHU Duct Mix Temp",
    presentValue=75, is_commandable=False
)

# create AHU return temp
_new_objects = analog_input(
    name="RA-T",
    properties={"units": "degreesFahrenheit"},
    description="AHU Duct Return Setpoint",
    presentValue=75, is_commandable=False
)

# create AHU supply fan VFD speed
_new_objects = analog_output(
    name="SF-O",
    properties={"units": "percent"},
    description="AHU Supply Fan VFD Speed",
    presentValue=75, is_commandable=True
)

# create AHU heating valve
_new_objects = analog_output(
    name="HTG-O",
    properties={"units": "percent"},
    description="AHU Heating Valve Command",
    presentValue=0, is_commandable=True
)

# create AHU cooling valve
_new_objects = analog_output(
    name="CLG-O",
    properties={"units": "percent"},
    description="AHU Cooling Valve Command",
    presentValue=0, is_commandable=True
)

# create AHU fan VFD run command
_new_objects = binary_output(
    name="SF-C",
    description="AHU Supply Fan Command",
    presentValue=True, is_commandable=True
)

# create AHU fan status input
_new_objects = binary_input(
    name="SF-S",
    description="AHU Supply Fan Status",
    presentValue=True, is_commandable=False,
)

# create AHU supply fan flow station
# OR can be totalized VAV box air flows
_new_objects = analog_input(
    name="SA-FLOW",
    properties={"units": "cubicFeetPerMinute"},
    description="AHU Duct Return Setpoint",
    presentValue=75, is_commandable=False
)

# create app
#bacnet = BAC0.lite(ip='10.0.2.20/24',deviceId='2021')
bacnet = BAC0.lite()

_new_objects.add_objects_to_application(bacnet)
bacnet._log.info("APP Created Success!")

def main():
    task1 = RecurringTask(changeValues, delay=30)
    task1.start()

    while True:

        time.sleep(10)

if __name__ == "__main__":
    main()

```

ChristianTremblay, 2023-03-05:
We need to talk...

I'm working on something.... not sure if I wish to make it public right now

bbartling, 2023-03-05:
Another idea I thought of is taking some trend logs of real data like a a
week's worth and replay it via a Pandas df.

Basically it's real data with no datetime index or time stamp and then when
the CSV file is read with Pandas you just calculate the time stamp into the
future from datetime.now() for each row into the future with
datetime.timedelta(). I use this to create fake datasets where it would be
really easy try this as well...DM me on LinkedIn else I'll try to post an
answer here to for this replay method.

bbartling, 2023-03-20:
@TShapinsky what does the `alfalfa-bacnet-bridge` do? Are there any examples to make fake HVAC equipment as a discoverable BACnet device?

I can only day dream of this but these points below are JCI naming conventions for AHU that contains all required input for ASHRAE Guideline 36 fault equations....where I think in theory G36 wants people to doing fault detection in the BAS which i have yet to see if any of the manufactures can pull this off or not. (am skeptical)

We have a few G36 research projects the company I work....the long story short it would be cool if there could be a tool to test this for the BAS manufactures or even IoT people wanting to FDD on the edge.

I used to work for JCI and need to cut over to a more brick or haystack naming convention!

```
# G36 AHU required input points
points = [
    "DAP-SP",
    "DAP-P",
    "SA-T",
    "SAT-SP",
    "MA-T",
    "RA-T",
    "SF-O",
    "HTG-O",
    "CLG-O",
    "SF-C",
    "SF-S",
    "SA-FLOW"]
```

TShapinsky, 2023-03-20:
@bbartling [Alfalfa](https://github.com/NREL/alfalfa) is a virtual building service which allows for realtime (or faster than realtime) interaction with energyplus or modelica models. Currently the points you get out are a function of the quality of the underlying model (there are some things which are exposed automatically, others need to be added manually). Alfalfa has limited haystack support with the intention of supporting brick in haystack (hopefully this year).

Alfalfa BACnet Bridge is a tool which allows you to connect an alfalfa simulation to a bacnet network. So far this has been tested with a Tridium BAS to demonstrate realtime control of a virtual building from the BAS interface.

bbartling, 2023-03-20:
@TShapinsky would there ever be some easy documentation on how to set Alfalfa up? A YouTube video would be amazing!

I haven't used simulations much at all like 99.9% of my experience has been setting up HVAC controls in the field so Modelica and energyplus is very new to me.

Would there be a IPython notebook a newbie could follow along like the 2 buildings example that includes the BACnet bridge for an AHU? Like this:
https://github.com/NREL/alfalfa-notebooks/blob/main/twobldgs.ipynb

ChristianTremblay, 2023-03-21:
> I used to work for JCI and need to cut over to a more brick or haystack naming convention!
> 
> ```
> # G36 AHU required input points
> points = [
>     "DAP-SP",
>     "DAP-P",
>     "SA-T",
>     "SAT-SP",
>     "MA-T",
>     "RA-T",
>     "SF-O",
>     "HTG-O",
>     "CLG-O",
>     "SF-C",
>     "SF-S",
>     "SA-FLOW"]
> ```

With Brick, Haystack or S223, you won't find a naming convention. It's more a matter of contextualizing information. Whatever the name is, it's related to A, B, C and has properties 1,2,3... it's probably a discharge air temperature...

IMO, it's one of the hardest challenge when trying to relate a controller app (aka the BACnet points) to a model.

bbartling, 2023-03-23:
I have the code in this repo of a bacnet server app to represent fake data:
https://github.com/bbartling/building-automation-helper/tree/master/fake_bacnet_device

It replays one of LBNLs public FDD datasets in a loop, every minute the BACnet app updates sensor values as read row by row in the CSV file as if its real data in real time. Its bit of a hack but it works for what I need. BACnet overriding heating valves or fan command points wont do anything but good enough for the moment of using this on a test bench for edge device FDD debugging..

Im going to close this issue for now I have something good enough for what I need it at the moment. Thanks all for responses!

closed 2023-03-23 by bbartling