ChristianTremblay/BAC0 #382: BAC0.core.io.IOExceptions.InitializationError

Ferkelcode, 2023-03-04:
Hi, 

I have used `bacnet = BAC0.connect(ip='134.xxx.xxx.x/24')` in my script, it shows error:
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (134.xxx.xxx.x) invalid. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask').

I have done several steps for troubleshooting:

1. ping 134.xxx.xxx.x, the IP is reachable.
2. checked port 47808: but there is nothing in the output. I have searched online, the reason could also be that I do not have permission to check. And I also was not able to run Command Prompt as a administrator. So I am not sure if the port 47808 is used.
3. Tried: bacnet=BAC0.lite(ip='134.xxx.xxx.x', port=**47809**), the error: [BAC0.core.io](http://bac0.core.io/).IOExceptions.InitializationError: IP Address provided (134.xxx.xxx.x:47809) invalid. Check if another software is using port **47808** on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask')
4. I changed another IP address (192.xxx.xxx.x), the error stays same.
5. I ran` netstat -ano -p tcp` , but there is not 47808 in the list. 

Could this be that the script has a problem, that even though my port has no problem, it will always trigger this error?
Thanks.

CrisCodesCrap, 2023-03-04:
Check UDP port 47808 instead of TCP

Ferkelcode, 2023-03-04:
> 

Thank you very much for your answer.
I tried: `netstat -ano -p udp`, but there is also no 47808 in the list. Does this mean the port 47808 is free for BACnet?

Ferkelcode, 2023-03-04:
> 

Thank you, I am using Windows.

I tried `netstat -ano | findstr :47808`, but there is no output. As this is the laptop from my university, could  it be that I don't have permission to check this?
[image]

CrisCodesCrap, 2023-03-04:
If it was regarding permissions it would have said so.

ChristianTremblay, 2023-03-05:
Is 134.x.x.x your local interface IP address ?
the term connect may mislead the user that could use the IP of the device he wishes to connect to...

closed 2023-05-21 by github-actions[bot]