ChristianTremblay/BAC0 #383: DCC_Time duration of disable

Chris-Wang-bjc, 2023-03-07:
hello, I used “Device Communication Control” service to disable the device, but the "timeDuration" argument didn't work.
Here is my code:
bacnet.dcc(address='address of device', duration=1, state='disable', password='password')

I am sure that all the parameters in this request are correctly configured and the request is sent correctly.
But when I checked the Wireshark logs, I didn't see the "timeDuration" parameter.
[image]
And the device failed to return to a working state after 1 minute.

Chris-Wang-bjc, 2023-03-08:
I already solved it. Please ignore it.

closed 2023-03-08 by Chris-Wang-bjc

ChristianTremblay, 2023-03-08:
And what what the problem ? I'm curious

Chris-Wang-bjc, 2023-03-08:
Sorry, this problem is due to other libraries I used, not BAC0, thanks!