ChristianTremblay/BAC0 #384: BAC0 can only discover virtual device from BOSS a Softdel BACner/IP Simulation System in the first few times

11208, 2023-03-22:
I have simulated BACNET using BOSS a Softdel BACner/IP Simulation System. 
I created a basic python app like
```
import BAC0

def main():
    bacnet = BAC0.connect(ip='192.168.1.156/24')
    bacnet.discover()
    print(bacnet.devices)

if __name__ == "__main__":
    main()
```

The first time I run the app device show up, 
[image]

but when I run it for the second time or third time no device show up.
[image]

11208, 2023-03-22:
I test with nmap and got this.
```
sudo nmap --script bacnet-info --script-args full=yes -sU -n -sV -p 47808 192.168.1.156
```
[image]

The is close but the simulator is still running. 
I also use Thingsboard Gateway to pull data and the data is save to db normally.
Why can BAC0 detect when the state is close.

11208, 2023-03-22:
Simulator issue

11208, 2023-03-22:
I tested with two BACnet simulators. One from SoftDEL and one from YABE AddOn

Step:
1. Start SoftDEL BOSS, Then call whois() and get response back.
2. Wait for a while and call whois again, empty response.
3. Start YABE Simulator Then call whois() and get response for SoftDEL and YABE back.

Seems like after a while the simulator don't listen or want to answer, unless another BACnet application start and trigger port 47808 or something.

Do we have any function that trigger BACNET to start answering whois or discover function again?

ChristianTremblay, 2023-03-23:
once BAC0 runs (and keep running) it acts as a BACnet device and will respond to I-Am ...

for your tests, just load a python interpreter and run your code. It will keep BAC0 running and it'll be easier to find what is going on

11208, 2023-03-23:
I did tried python interpreter and only able to get the response when calling discover() or whois() for a few times after the Simulator started then getting empty response as before. May I have to find a work around to get device list when calling those two function after time passed.

My intension is to query a device list and its properties using BAC0 when function call.

11208, 2023-03-23:
Seems like my issue is related to #331

11208, 2023-03-29:
Here is the screen shot of wireshark.
[image]

Does anyone have an idea on how to fix this?

bbartling, 2023-03-29:
I never have issues with using BAC0 but I am always in the field on real BACnet devices on real hardware where there is no VM's or networking jerry rigging to accommodate running multiple BACnet devices on one device.

Something like this always work for me on real equipment:

```
import BAC0, time

# define bacnet app
bacnet = BAC0.lite()
time.sleep(2)

# generates all BACnet addresses some are not VAV boxes
devices = bacnet.whois(global_broadcast=True)

for device in devices:
    if isinstance(device, tuple):
        device_mapping[device[1]] = device[0]
        print("Detected device %s with address %s" % (str(device[1]), str(device[0])))
print(device_mapping)
print((str(len(device_mapping)) + " devices discovered on network."))
```

closed 2023-06-05 by github-actions[bot]