ChristianTremblay/BAC0 #386: ERROR   | Super Mega Giga big error 'NoneType' object has no attribute 'properties'. Removing task

marganellim, 2023-04-12:
Having a strange issue with BAC0 stopping its polling cycle for some reason and I cant seem to identify where the problem is coming from. 

I have a simple example here that reads 5 points from 2 devices and writes the results to influxDB. 
This seems to work great for about 10 minutes. 
After about 10 minutes, I get the following error from BAC0 and values are no longer saved to influx. 

Here is the example script.

```
import BAC0
from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
)
import time
import config

_params = {
    "name" : "InfluxDB",
    "url" : config.influx_url,
    "port" : config.influx_port,
    "token" : config.influx_token,
    "org" : config.influx_org,
    "bucket" : config.influx_bucket
    }

bacnet = BAC0.connect(ip=config.device_ip,deviceId=config.device_address,port=47809,localObjName=config.device_name,description="",db_params=_params)

config_devices = config.devices

for cdev in config_devices:
    points_list = cdev["points"]
    dev_add = cdev["address"]
    dev_ip = cdev["ip"]
    try:
        BAC0.device(dev_ip,dev_add,bacnet,object_list=points_list,poll=120)
    except Exception as e:
        print(e)
        print("Failed to get device")
        quit()

while True:
    time.sleep(1)
```

Here is the output from BAC0
```

$ python3 readPoints_influx.py 
2023-04-12 10:49:22,460 - INFO    | Starting BAC0 version 22.9.21 (Complete)
2023-04-12 10:49:22,460 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-04-12 10:49:22,461 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-04-12 10:49:22,461 - INFO    | Starting TaskManager
2023-04-12 10:49:22,461 - INFO    | Using ip : 192.168.3.110:47809
2023-04-12 10:49:22,467 - INFO    | Starting app...
2023-04-12 10:49:22,467 - INFO    | BAC0 started
2023-04-12 10:49:22,467 - INFO    | Registered as Simple BACnet/IP App
2023-04-12 10:49:22,469 - INFO    | Update Local COV Task started
2023-04-12 10:49:22,475 - INFO    | Connection made to InfluxDB bucket : TEST
2023-04-12 10:49:23,555 - INFO    | Server started : http://192.168.3.110:8111
 * Serving Flask app 'BAC0.web.FlaskServer' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:8111
 * Running on http://192.168.3.110:8111 (Press CTRL+C to quit)
2023-04-12 10:49:27,557 - INFO    | Found those networks : set()
2023-04-12 10:49:27,557 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
2023-04-12 10:49:30,560 - INFO    | Changing device state to DeviceDisconnected'>
2023-04-12 10:49:30,565 - INFO    | Changing device state to RPDeviceConnected'>
2023-04-12 10:49:30,572 - INFO    | Device 12344:[iSMA-B-4U4A-H-IP] found... building points list
2023-04-12 10:49:30,581 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2023-04-12 10:49:30,589 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2023-04-12 10:49:30,598 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2023-04-12 10:49:30,607 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2023-04-12 10:49:30,607 - INFO    | Ready!
2023-04-12 10:49:30,607 - INFO    | Device defined for normal polling with a delay of 120sec
2023-04-12 10:49:30,607 - INFO    | Polling started, values read every 120 seconds
2023-04-12 10:49:30,607 - INFO    | Changing device state to DeviceDisconnected'>
2023-04-12 10:49:30,612 - INFO    | Changing device state to RPMDeviceConnected'>
2023-04-12 10:49:30,619 - INFO    | Device 123445:[SBIO BACnet Engine] found... building points list
2023-04-12 10:49:30,622 - INFO    | Ready!
2023-04-12 10:49:30,622 - INFO    | Device defined for normal polling with a delay of 120sec
2023-04-12 10:49:30,622 - INFO    | Polling started, values read every 120 seconds
2023-04-12 11:01:34,351 - ERROR   | Super Mega Giga big error 'NoneType' object has no attribute 'properties'. Removing task.
2023-04-12 11:03:31,129 - ERROR   | Super Mega Giga big error 'NoneType' object has no attribute 'properties'. Removing task.

```

My polling rate is 120 seconds for both devices and I've tried adjusting the polling rate both up  and down. 
When the polling rate is 60 seconds the error occurred after 7 minutes. 
When the polling rate is set to 30 seconds, the error occurred after 3 minutes. 
When the polling rate is set to 10 seconds, the error occurred after 1 minute

I am not sure what's throwing the error but it is consistent. 
Is there a way to detect when this error occurs in the script so I can take action, and is there a way to narrow down what is throwing this problem?

Thanks

bbartling, 2023-04-12:
What happens if you try just the `bacnet = BAC0.connect(ip=config.device_ip)`
on the device without the database stuff just to see if you get the same
error?

I've only seen that error when using the recurring task:
`from BAC0.tasks.RecurringTask import RecurringTask`
```
def main():
    # run every second to check value
    task1 = RecurringTask(fc.update, delay=1)
    task1.start()

    while True:

        time.sleep(10)

if __name__ == "__main__":
    main()
```

marganellim, 2023-04-12:
It looks like that does solve the problem. Removing the influxDB extras, the script runs for 10 minutes without the same error occurring. 
Does this indicate there's a problem with my influx configuration or is it possible the influx option is not 100%?

bbartling, 2023-04-12:
Great question...I actually never got that far in dabbling with a database,
I actually dont know much about them.

For troubleshooting purposes could you ever create a Python script that can
manually add data to the db, like simulate your BACnet data transfer?

marganellim, 2023-04-12:
Yeah if the influxdb functionality isn't working I'll read the points manually on a poll and write them myself. That's not too bad. Was hoping BAC0 would take care of it automagically, but that's not the end of the world. 

Thanks for the quick response !

bbartling, 2023-04-12:
well I think its supposed to care of it! I am interested in learning the db features of BAC0 someday!

marganellim, 2023-04-12:
Just an update, this problem was not related to the influxdb configuration as far as I can see. I put the influx stuff back and so far things are working perfectly. I rebuilt my influx server, so its possible there was a mis-configuration there. 
It would be nice to be able to check somehow that the logging has stopped or have the software throw a significant error that I can test for

bbartling, 2023-04-13:
Nice to know its working....would you know of any good YouTube videos or tutorials on how to setup Influx? And recreate what you have been doing?

Also is there anywhere in the docs that could be expanded on for directions on setting up BAC0+db for the newb?

closed 2023-06-21 by github-actions[bot]

ohshazbot, 2023-11-13:
@bbartling Former colleague had this little sandbox for spinning up influx in a local container, if you just need something for toying with https://github.com/gnmerritt/influxdb-sandbox-1x