ChristianTremblay/BAC0 #387: bufferOverflow when reading bacnet_properites

DinosaurierRex, 2023-04-18:
Hello,
Im currently running a script to read out all endpoints of a specific controller. To find my error i tried out a minimalistic approach like

```
>>> import BAC0
>>> bacnet = BAC0.lite(port=46000)
>>> bacet.whois()    # This returns a list of 7 devices like [('0x1bf90201be00', 300), ... ]

# Then i try to connect to device
>>> temp_device = temp_device = BAC0.device('0x1bf90201be00', 300, bacnet)

# Reading the points list also works
>>> temp_device.points   # This seems to return a list of all points in following format -> [objectName : presentValue engineeringUnits, ... ]

# However trying to read the properties doesnt work
>>> temp_device.bacnet_properties

WARNING | APDU Abort Reject Reason : bufferOverflow
WARNING | No response from controller bufferOverflow
```

Some of the other devices work fine, some get a bufferOverflow. Is there a known solution?

Thanks in advance :smile:

ChristianTremblay, 2023-05-05:
Not that I know for sure.
Looks like a device that cannot handle a lot of information at the same time. Fills its memory and boom.

Maybe you could try using read range ?

DinosaurierRex, 2023-06-02:
Thanks for your help :)

I managed to deal with it by iterating over the units and states. Somehow the error only occured when accessing the `device.bacnet_properties` of a connected device. If using `device.read_property("modelName)` it just works fine. Same goes for analog units or binary states.

ChristianTremblay, 2023-09-09:
I do my development with mostly JCI BACnet controllers.... which means that some use case are missed when devices have less memory (which looks to de be the case...)

This means that the properties fucntion should have some kind of backup mechanism to switch to simple read when read_multiple fails... something to think about.

history_size is local to BAC0 and refer to the number of record we keep in memory when we use `BAC0.device`

closed 2023-10-30 by DinosaurierRex