ChristianTremblay/BAC0 #388: Issue running two connections at once

rcb-tcs, 2023-05-01:
Hello, I am still noticing the same issues mentioned in #265 with more recent versions of BAC0 (22.9.21) and bacpypes (0.18.6) 
 running the following 3-line script:

```
import BAC0
bacnet1 = BAC0.lite('192.168.1.17/24:47808')
bacnet2 = BAC0.lite('192.168.1.17/24:47809') # different port!
```
I ran this script lots of times, and further determined that:
- it's timing dependent. I see several different errors (including the one in the mentioned in the previous issue). Sometimes there is no error.
- it's transient. If I add code afterwards to use the connections or dump into pdb, both connections behave exactly as expected.

I looked into the code a bit to try to figure out what's going on, and best I can tell what is happening is that when the `bacnet2` connection is being set up, the `Base._startAppThread()` thread created while starting `bacnet1` is trying to process deferred work from the `bacnet2` main thread is finished setting it up. I found the following hack totally fixes the error and leaves me with (apparently) fully operational `bacnet1` and `bacnet2` connections.
```
import BAC0
import bacpypes.core
import time

bacnet1 = BAC0.lite('192.168.1.17/24:47808')
bacpypes.core.running = False
bacnet1.t.join()
bacnet2 = BAC0.lite('192.168.1.17/24:47809')
```
I'm not sure what the best way to deal with this issue is... I may not be understanding it correctly, but it almost seems to me as if the "run" thread created for each separate link should be made a singleton like BAC0 TaskManager, since the `bacpypes.core.run()` function doesn't have any actual link its BAC0Application "owner". As of now, it seems as if there is no mechanism to prevent the work deferred by `bacnet2` from being picked up by the thread owned by `bacnet1` and visa versa.

Thanks!

ChristianTremblay, 2023-05-05:
Thanks for that. 

It is really time dependent as you point. For example, something like this

```
import BAC0
bacnet1 = BAC0.lite('192.168.1.17/24:47808')
bacnet2 = BAC0.lite('192.168.1.17/24:47809') # different port!
```

works perfectly... So is this just a matter of waiting for the socket to be registered before trying to re-register another port on the same adapter ?

Now... what do I do with this ?

I tried to reduce the delay... just to see if there was a threshold and I can see that a very short wait is sufficient

```
In [1]: import BAC0
   ...: import time
   ...: bacnet1 = BAC0.lite('192.168.211.16/24:47808')
   ...: time.sleep(0.001)
   ...: bacnet2 = BAC0.lite('192.168.211.16/24:47809') # different port!
2023-05-04 20:50:47,170 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2023-05-04 20:50:47,170 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-05-04 20:50:47,171 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-05-04 20:50:47,171 - INFO    | Starting TaskManager
2023-05-04 20:50:47,171 - INFO    | Using ip : 192.168.211.16
2023-05-04 20:50:47,275 - INFO    | Starting app...
2023-05-04 20:50:47,276 - INFO    | BAC0 started
2023-05-04 20:50:47,276 - INFO    | Registered as Simple BACnet/IP App
2023-05-04 20:50:47,293 - INFO    | Update Local COV Task started
2023-05-04 20:50:47,294 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2023-05-04 20:50:47,294 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-05-04 20:50:47,295 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-05-04 20:50:47,295 - INFO    | Using ip : 192.168.211.16:47809
2023-05-04 20:50:47,296 - INFO    | Starting app...
2023-05-04 20:50:47,297 - INFO    | BAC0 started
2023-05-04 20:50:47,297 - INFO    | Registered as Simple BACnet/IP App
2023-05-04 20:50:47,308 - INFO    | Update Local COV Task started

In [2]: exit
PS C:\Users\ctremblay> ipython
Python 3.11.3 (tags/v3.11.3:f3909b8, Apr  4 2023, 23:49:59) [MSC v.1934 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.13.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import BAC0
   ...: import time
   ...: bacnet1 = BAC0.lite('192.168.211.16/24:47808')
   ...: time.sleep(0.0001)
   ...: bacnet2 = BAC0.lite('192.168.211.16/24:47809') # different port!
2023-05-04 20:51:05,074 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2023-05-04 20:51:05,074 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-05-04 20:51:05,074 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-05-04 20:51:05,074 - INFO    | Starting TaskManager
2023-05-04 20:51:05,074 - INFO    | Using ip : 192.168.211.16
2023-05-04 20:51:05,176 - INFO    | Starting app...
2023-05-04 20:51:05,176 - INFO    | BAC0 started
2023-05-04 20:51:05,177 - INFO    | Registered as Simple BACnet/IP App
2023-05-04 20:51:05,185 - INFO    | Update Local COV Task started
2023-05-04 20:51:05,186 - INFO    | Starting BAC0 version 22.9.21 (Lite)
2023-05-04 20:51:05,186 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-05-04 20:51:05,186 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-05-04 20:51:05,186 - INFO    | Using ip : 192.168.211.16:47809
2023-05-04 20:51:05,187 - INFO    | Starting app...
2023-05-04 20:51:05,187 - INFO    | BAC0 started
2023-05-04 20:51:05,187 - INFO    | Registered as Simple BACnet/IP App
2023-05-04 20:51:05,188 - INFO    | Update Local COV Task started

an error has occurred: 'UDPDirector' object has no attribute 'request'
Traceback (most recent call last):
  File "C:\Users\ctremblay\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\bacpypes\core.py", line 154, in run
    asyncore.loop(timeout=delta, count=1)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncore.py", line 212, in loop
    poll_fun(timeout, map)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncore.py", line 137, in poll
    is_w = obj.writable()
           ^^^^^^^^^^^^^^
  File "C:\Users\ctremblay\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\bacpypes\udp.py", line 231, in writable
    return (not self.request.empty())
                ^^^^^^^^^^^^
AttributeError: 'UDPDirector' object has no attribute 'request'
In [2]:
```

ChristianTremblay, 2023-05-05:
And to add to confusion, Windows and Linux don't behave the same...

closed 2023-07-12 by github-actions[bot]