ChristianTremblay/BAC0 #389: Send BAC0.lite object as parameter to multiprocessing

coolrhot, 2023-05-08:
Hi,

We are trying to setup a system where we are sending multiple read requests to multiple BACnet servers from a single BAC0 client. 

So to achieve that was planning to use a single BAC0 lite client object and send it to multiple processes created via multiprocessing.ProcessPoolExecutor. 

I was not able to send the BAC0 object as argument to the process then later tried to keep the BAC0 object as global variable and accessing it in the process. But for some reason once the BAC0 object is used to read inside the process it gets held indefinitely. Works if the object is used outside the process.

Later tried to just use discover call inside the process but still it did not complete the process.

import BAC0
import multiprocessing

test = None
test = BAC0.lite('127.0.0.1', '47808')

def run():
    global test
    test.discover()
    print(type(test))

process = multiprocessing.Process(target=run)
process.start()

Can you please suggest if something should be tried in an alternative way or if this is possible.

bbartling, 2023-05-08:
Hi I am not sure about multiprocessing but I use a script to make a lot of BACnet reads and writes in a synchronous fashion all the time something like this:
https://gist.github.com/bbartling/88e95557b0b297bea721c0f285657d53

This is a typical ordinary office building where the computer running these scripts are all right along side the BACnet building automation system on all on the same subnet.

Can you tell me about the multiprocessing part, what is that used for? Like an app the is doing something else in the background and then periodically some BACnet tasks? Can you run it all in a synchronous fashion together?

bbartling, 2023-05-08:
From my experience BAC0 runs real nice inside Flask apps and there is a "RecurringTask" feature that uses threading under the hood to do recurring tasks on time intervals

ChristianTremblay, 2023-05-08:
I never tried multiprocessing. 

The way BAC0 is made, it needs to capture the socket and keep it open, on the port chosen for the time of the script. 

Because BAC0 becomes a BACnet device on the network, like any other device. It must be kept alive and ready to answer. 

For that reason, I suspect it will be complicated to rely on multi processes. 

I would try using "read multiple" requests to optimize traffic instead of simple read. 

Still in beta, but bacpypes3 support async io, maybe it would be faster for that use case.

coolrhot, 2023-05-09:
Thank you for the quick response.

@bbartling 
We are setting up an environment where there are several BACnet devices will be present which will all be updated at a specific point of time. So was hoping to read the values using a BACnet client at one instant.

To read all the devices at the same time was planning to use multiprocessing to create one process for each BACnet device and use a same single BACO client to read from each device. 

To read from each of the devices simultaneously was planning to send the BAC0.lite object to each of the process and then use read or readMultiple command from that specific process to the corresponding BACnet device.

I'm doubting that once the BAC0 object is used in the created process it is waiting at 
File "/usr/local/lib/python3.7/dist-packages/bacpypes/iocb.py", line 149, in wait
    return self.ioComplete.wait(*args, **kwargs)

@ChristianTremblay 
Async IO will definitely be useful will check that out too.

closed 2023-07-16 by github-actions[bot]