ChristianTremblay/BAC0 #392: Trying to detect a BAC0 virtual device using yabe

KijanDaniel, 2023-05-24:
I tried to create a virtual bacnet device (in this case a thermometer) using BAC0.lite and trying to detect it using yabe. It says that I cannot make a connection on yabe because the port is already taken. Is there any way you could help me with this. I have been following your code examples but still could not solve the problem. I even used BAC0.device but from my understanding this will take up a port because it is just trying to listen to a device. Thankyou in advance.

bbartling, 2023-05-24:
try it from a different computer on the same LAN...else I think you are going to have to run one BACnet device on a different port.

@ChristianTremblay can you run a BAC0 app on port 47809 and then use what ever BACnet scan tool on the same computer where I think they default to port designated for BACnet port 47808?

ChristianTremblay, 2023-05-24:
My preferred way is to define multiple IP addresses on the same network interface.

In windows, you go this way
[image]

Then, you can attach BAC0 to one IP...and probably Yabe to the other one.

I do this all the time. 

Using different ports add a layer of complexity as you have to play with broadcasts subtilities. And I actually know that Windows and Linux do not operate the same sharing the same IP with different ports.... BAC0 doesn't like it

closed 2023-06-02 by KijanDaniel