ChristianTremblay/BAC0 #396: What is the value format when write property statusFlags?

saimadao, 2023-05-29:
I knonw the statusFlags is read only. But as a tester I need to write it then get a writeAccessDenied to verify it's read only.

I have tried 
`myBACnet.write('192.168.1.10 analogValue 0 statusFlags {"in_alarm":"False","fault":"False","overridden":"False","out_of_service":"False"}')`
and 
`myBACnet.write('192.168.1.10 analogValue 0 statusFlags {"in_alarm":False,"fault":False,"overridden":False,"out_of_service":False}')`
and 
`myBACnet.write('192.168.1.10 analogValue 0 statusFlags {"in_alarm":0,"fault":0,"overridden":0,"out_of_service":0}')`
and 
`myBACnet.write('192.168.1.10 analogValue 0 statusFlags [0,0,0,0]')`
all of them get **TypeError: invalid constructor datatype**

Could anyone tell me what should I do to get **writeAccessDenied** exception when write statusFlags?

saimadao, 2023-05-30:
@bbartling,  in BTL certification require testing each properties' write-read access. and I want to automat this verification process. 
So I use BAC0 send write command to each property, then according each property's write-read access in requirement, to assert the write operation is success or return a writeAccessDenied exception.
I write with BAC0 to all properties with no problem, except property **statusFlags**. 
I read this property and get [0,0,0,0] returned. but when I tried to write the same value [0,0,0,0] to **statusFlag** I got **TypeError: invalid constructor datatype** it seems BAC0 doesnot support the value type of list like [0,0,0,0]. besides I tried 0,0,0,0 and other combinations in my original question also got the same error.

bbartling, 2023-05-30:
@saimadao that is really neat, thanks this is just for my own learning purposes. I've only used BACnet in the field to make HVAC systems "do things" with read/write presentValue of setpoints, etc..

Hope fully someone can answer your questions...I think it would be related to bacpypes ?

ChristianTremblay, 2023-05-30:
Interesting use case....
You probably need to build a custom request to be sent. 
Usually, status flags are set up internally in BAC0 using the "python" list ([0,0,0,0]) and as it is technically a read only thing, I never built a specific request to write to a status flag.

I'll make a few verif...but it is a pretty edgy/rare use case.

ChristianTremblay, 2023-06-22:
nope. Not hing I can remember

closed 2023-08-28 by github-actions[bot]