ChristianTremblay/BAC0 #397: How to write the Value of property "state-text" with parameter "array index"?

Chris-Wang-bjc, 2023-05-31:
Hi, all
I wanna to write the value of property "state-text", when writing the value, I need to pass the "array index" parameter, however, the currently provided function does not seem to support passing this parameter. So I want to know, when I write this value, how can I pass this parameter in? Or whether I can customize the parameters of this function?
[image]

Chris-Wang-bjc, 2023-05-31:
I want the function "write" to be able to pass in the arguments that I need, thanks

bbartling, 2023-05-31:
what error do you get?

Chris-Wang-bjc, 2023-06-01:
No error, I'm just wondering how I can pass the parameter "array index" into the write function.

Chris-Wang-bjc, 2023-06-01:
This may not be an issue, but I have no way to contact you and can only ask you some questions in this way.

Chris-Wang-bjc, 2023-06-01:
I have solved it, so I close this issue, thanks very much!

closed 2023-06-01 by Chris-Wang-bjc

ChristianTremblay, 2023-06-01:
It is perfectly fine to ask questions this way. 

What would be very nice would be you post the answer you found. This way, other will find it if they ever look for it. 

And eventually, it gives some "documentation" material. 

Thanks