ChristianTremblay/BAC0 #399: How can I connect to my virtual bacnet server if I made it in another container

KijanDaniel, 2023-06-06:
I made a bacnet server using the .connect function in one container. I want to be able to connect a device I made in another container but I cannot because the .device function needs the bacnet variable from the server. Is there a workaround for this?

bbartling, 2023-06-06:
If you figure out how to do BACnet in containers can you make a YouTube video?

KijanDaniel, 2023-06-07:
@bbartling I can explain it to you I got it to work in one container

KijanDaniel, 2023-06-07:
import pytest
import time
from collections import namedtuple
import bacpypes
import socket
import random
import pickle
from bacpypes.object import (
    AnalogInputObject,
    AnalogValueObject,
    BinaryValueObject,
    Property,
    register_object_type,
    registered_object_types,
    DatePatternValueObject,
    ReadableProperty,
    WritableProperty,
    OptionalProperty,
    CharacterStringValueObject
)
import BAC0
from BAC0.core.proprietary_objects.jci import tec_short_point_list
from BAC0.core.devices.local.object import ObjectFactory
from BAC0.core.devices.local.models import make_state_text

from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
    binary_input,
    binary_output,
    binary_value,
    multistate_input,
    multistate_output,
    multistate_value,
    date_value,
    datetime_value,
    temperature_input,
    temperature_value,
    humidity_input,
    humidity_value,
    character_string,
)

my_obj_list = [('file', 1),
             ('analogInput', 2),
             ('analogInput', 3),
             ('analogInput', 5),
             ('analogInput', 4),
             ('analogInput', 0),
             ('analogInput', 1)]

import jsonpickle

hostname = socket.gethostname()

localip = socket.gethostbyname(hostname)

host= localip #client ip
port = 4005
    
server = (localip, 4000)
    
s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.bind((host,port))

s.sendto(hostname.encode('utf-8'), server)

bacnet, addr = s.recvfrom(102400)

bacnet = jsonpickle.decode(bacnet.decode('utf-8'))

print(bacnet)

tempSensor = BAC0.lite(ip=localip,port='47810')

print(tempSensor)

#bbmdAddress=localip+':47809', bbmdTTL=900

tempratureChange = 70
_new_objects = analog_output(presentValue=89.9, is_commandable=True) #This gives the tempSensor two values we can play with 
_new_objects = analog_value(presentValue=tempratureChange, is_commandable=True)

_new_objects.add_objects_to_application(tempSensor) #This assigns the objects (aka points) to the tempSensor

ip = tempSensor.localIPAddr.addrTuple[0] #This gives us the Ip address needed in the connect function
boid = tempSensor.Boid #This gives us the Boid which is also needed in the connect function
#print("{}/14:47809".format(ip))
connectTempSensor = BAC0.device("{}:47809".format(ip),boid,bacnet,object_list = my_obj_list,poll=10)

while True:
    time.sleep(2)
    tempratureChange = random.random()

KijanDaniel, 2023-06-07:
Thats the node

KijanDaniel, 2023-06-07:
import pytest
import time
from collections import namedtuple
import bacpypes
import socket
from bacpypes.object import (
    AnalogInputObject,
    AnalogValueObject,
    BinaryValueObject,
    Property,
    register_object_type,
    registered_object_types,
    DatePatternValueObject,
    ReadableProperty,
    WritableProperty,
    OptionalProperty,
    CharacterStringValueObject
)
import BAC0
from BAC0.core.proprietary_objects.jci import tec_short_point_list
from BAC0.core.devices.local.object import ObjectFactory
from BAC0.core.devices.local.models import make_state_text

from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
    binary_input,
    binary_output,
    binary_value,
    multistate_input,
    multistate_output,
    multistate_value,
    date_value,
    datetime_value,
    temperature_input,
    temperature_value,
    humidity_input,
    humidity_value,
    character_string,
)

import socket
import jsonpickle

hostname = socket.gethostname()

localip = socket.gethostbyname(hostname)

print("localip: " + localip)

bacnet = BAC0.lite(ip=localip, port='47809')

#bbmdAddress=localip+':47808', bbmdTTL=900,

pickleBacnet = jsonpickle.encode(bacnet)

def Main():
    
    name = socket.gethostname()
    host = socket.gethostbyname(name)
    port = 4000

    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    s.bind((host, port))

    print("Server Started")
    while True:
        data, addr = s.recvfrom(102400)
        s.sendto(pickleBacnet.encode('utf-8'), addr)
    c.close()

Main()

ChristianTremblay, 2023-06-09:
I highly suspect the `sleep` call... BAC0 needs to be available all the time, by sleeping 2 seconds, you pause everything.

Here is an example of how I deal with internal present value updates
it is an extract of the code, it's incomplete, the goal is just to show 

```python
def update(obj, timestamp, flags):
    _reference_datetime = datetime.fromtimestamp(timestamp).astimezone(pytz.UTC)
    _local_datetime = _reference_datetime.astimezone()
    _reference_date = Date(_local_datetime.date().isoformat())
    _reference_time = Time(_local_datetime.time().isoformat())
    if flags == [0, 0, 0, 0]:
        app.dev[obj].presentValue.time = _reference_time
        app.dev[obj].presentValue.date = _reference_date
    # print(_reference_date)
    app.dev[obj].statusFlags = flags

def main():
    update()
    task_device = RecurringTask(update, delay=int(os.environ.get("OWM_REFRESHRATE")))
    task_device.start()
    while True:
        time.sleep(0.001)

```

ChristianTremblay, 2023-06-22:
First, stop using `time.sleep` for more than a few fractions of ms.. it's evil and will make BAC0 cry. You don't want BAC0 crying...

Name your points. It makes your life easier.

This code is untested, but you should be close enough

```python
import pytest
import time
from collections import namedtuple
import bacpypes
import socket
from bacpypes.object import (
    AnalogInputObject,
    AnalogValueObject,
    BinaryValueObject,
    Property,
    register_object_type,
    registered_object_types,
    DatePatternValueObject,
    ReadableProperty,
    WritableProperty,
    OptionalProperty,
    CharacterStringValueObject
)
import BAC0
import time
import random
from BAC0.core.devices.local.models import make_state_text

from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
    binary_input,
    binary_output,
    binary_value,
    multistate_input,
    multistate_output,
    multistate_value,
    date_value,
    datetime_value,
    temperature_input,
    temperature_value,
    humidity_input,
    humidity_value,
    character_string,
)

name = socket.gethostname()

ip = socket.gethostbyname(name)

device = BAC0.lite(ip=ip, deviceId= "1",port="47809",bbmdAddress=ip+':47809', bbmdTTL=900)
_new_objects = analog_input(name="myTempSensor", presentValue = varyingTemp) #a sensor is an input...not an AO
_new_objects = analog_value(name="aSetpoint", presentValue=90, is_commandable=True)
_new_objects.add_objects_to_application(device) #This assigns the objects (aka points) to the BACnet device

def update():
    device["myTempSensor"].presentValue = random.random() * 100
    device["myTempSensor"].statusFlags = [0,0,0,0]
    
def main():
    update()
    task_device = RecurringTask(update, delay=10)
    task_device.start()
    while True:
        time.sleep(0.001)
```

closed 2023-08-28 by github-actions[bot]