ChristianTremblay/BAC0 #400: The COV function is not working properly

KijanDaniel, 2023-06-14:
I am running the cov function and it runs okay but it keeps printing random errors, but it is not crashing. What is going on here.

The error:
exception: AttributeError("'NoneType' object has no attribute 'cov_registered'")
2023-06-14 10:09:27 Traceback (most recent call last):
2023-06-14 10:09:27   File "/usr/local/lib/python3.10/site-packages/bacpypes/app.py", line 382, in indication
2023-06-14 10:09:27     helperFn(apdu)
2023-06-14 10:09:27   File "/usr/local/lib/python3.10/site-packages/BAC0/core/app/ScriptApplication.py", line 129, in do_ConfirmedCOVNotificationRequest
2023-06-14 10:09:27     self.subscription_contexts["context_callback"](elements)
2023-06-14 10:09:27   File "/usr/local/lib/python3.10/site-packages/BAC0/core/functions/cov.py", line 149, in context_callback
2023-06-14 10:09:27     device[elements["object_changed"]].cov_registered = True

ChristianTremblay, 2023-06-22:
I need more info on that...

KijanDaniel, 2023-06-27:
@ChristianTremblay This is my device code:
import pytest
import time
from collections import namedtuple
import bacpypes
import socket
from bacpypes.object import (
    AnalogInputObject,
    AnalogValueObject,
    BinaryValueObject,
    Property,
    register_object_type,
    registered_object_types,
    DatePatternValueObject,
    ReadableProperty,
    WritableProperty,
    OptionalProperty,
    CharacterStringValueObject
)
import BAC0
import time
import random
from BAC0.core.proprietary_objects.jci import tec_short_point_list
from BAC0.core.devices.local.object import ObjectFactory
from BAC0.core.devices.local.models import make_state_text

from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
    binary_input,
    binary_output,
    binary_value,
    multistate_input,
    multistate_output,
    multistate_value,
    date_value,
    datetime_value,
    temperature_input,
    temperature_value,
    humidity_input,
    humidity_value,
    character_string,
)

name = socket.gethostname()

ip = socket.gethostbyname(name)

device = BAC0.connect(ip=ip, deviceId= "1",port="47809",bbmdAddress=ip+':47809', bbmdTTL=900)

varyingTemp = 90
_new_objects = analog_output(presentValue = varyingTemp, is_commandable=True) #This gives the tempSensor two values we can play with 
_new_objects = analog_value(presentValue=90, is_commandable=True)

_new_objects.add_objects_to_application(device) #This assigns the objects (aka points) to the tempSensor

while True:
    varyingTemp = random.random() * 100
    
    device["AO"].presentValue = varyingTemp
    time.sleep(10)

KijanDaniel, 2023-06-27:
@ChristianTremblay  I am getting an error from your code that says the above error still. It appears to be an issue in your code. Could you verify this or give me a fix.

KijanDaniel, 2023-06-27:
@ChristianTremblay It appears that the device in your function (def context_callback(self, elements):) in cov.py the device does not have the elements that you are referencing because it has none type

KijanDaniel, 2023-06-27:
@ChristianTremblay I commented out a large majority of that function and it seems to work just fine is that an okay work around?

closed 2023-09-03 by github-actions[bot]